realsenseai/librealsense #8407: C# Tutorial-4 is failing

bobdavies2000, 2021-02-21:
D455 camera
Latest firmware
Windows 10
SDK - latest
language is C#

### Issue Description
When I attempt to run the C# tutorial-4, it fails with the error 'Error while running Colorizer processing block, check the log for info'.  There is no additional info in the log.

The other 3 C# tutorials run fine.  I have made no changes to the librealsense install build.  

Bob Davies

MartyG-RealSense, 2021-02-21:
Hi @bobdavies2000   As a starting point in investigating your case, may I first ask please whether you may have installed on your Windows computer the RealSense **UWP** (Universal Windows Platform) driver, especially version **6.11.160.21** of it?

[image]

THe UWP driver is not required to be installed in order to use RealSense cameras with Windows.  It enables RealSense to interface with products that have UWP support, such as the Microsoft Hololens headset.  There have been a small number of reports of this particular version of the UWP driver causing errors with the C# wrapper when installed thoough.  The earlier **6.11.160.14** version of the UWP driver apparently doe not cause these errors.

bobdavies2000, 2021-02-22:
Marty - thanks for the note.  I installed the DMFT that you recommended from here:

https://downloadcenter.intel.com/download/29827/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

and it said it was installed properly but the tutorial-4 still fails.  I plugged/unplugged the camera and even rebooted to make sure but it is still failing with the same error running the colorizer processing block.  My Windows version 19041.804 on an Intel Core i9-7980XE and the librealsense is the one released on GitHub 2/4/21.  The camera is a D455.  Do you NOT see the problem with Tutorial-4?  This could be interesting to figure out what is different.

Bob Davies

MartyG-RealSense, 2021-02-22:
I was asking if you already had version **6.11.160.21** of the UWP driver already installed in order to determine if it was a possible cause of your problem, because I would recommend **not** installing that driver version.  It does not seem to have negatively affected your particular RealSense C# setup though.

I have not had any past reports of problems with the **cs-tutorial-4-software-dev** example.  Could you provide details of the error message that you receive please?

bobdavies2000, 2021-02-22:
Marty - the error is "Error while running Colorizer processing block, check the log for info" but there is nothing in the log.  I set the LRS_LOG_LEVEL to DEBUG, restarted Visual Studio, and there was still nothing.  Or perhaps the log is in some file.  I could not find the name of any file in my web searches.

Bob

MartyG-RealSense, 2021-02-22:
Do you have any other RealSense camera models other than D455 to test the example with?

bobdavies2000, 2021-02-22:
Tutorial-4 is failing for a D435i attached to my Core i9 system.  I tested on a Core i7 system with the upgraded D455 and tutorial-4 works (as do both a D435i and a second D455 camera.)  I never upgraded the second D455.  This would indicate that the problem is on the host side, not the camera side.  The second system is running the same version of Windows (the latest) but had an older version of librealsense - from late September.  I installed the latest version of librealsense and tutorial-4 is failing but with a different message - just "System.InvalidOperationException" in the RealSensed.dll - even with the LRS_LOG_LEVEL set to DEBUG.  I have kept the older version of librealsense that was working so I should be able to toggle the problem if you have further recommendations.

bobdavies2000, 2021-02-22:
One more update: I tested the other D455 with firmware 05.12.09.00 on my Core i9 system and it is failing with the same message -  "Error while running Colorizer processing block, check the log for info".

MartyG-RealSense, 2021-02-23:
I have referred this issue to Intel for advice.  Thanks very much for you patience.

bobdavies2000, 2021-02-23:
Marty - I assumed you worked for your Intel!  Fortunately, I am not in a hurry to solve this problem.

Bob

MartyG-RealSense, 2021-02-23:
I do work for Intel.  :)  I mean that I submitted a request for advice internally.

MartyG-RealSense, 2021-02-24:
Hi again @bobdavies2000  A minor bug was identified by Intel and a fix submitted in https://github.com/IntelRealSense/librealsense/pull/8428 - thanks for highlighting the issue!

closed 2021-02-24 by bobdavies2000

maloel, 2021-03-25:
@bobdavies2000 
We've merged the PR into our development branch, along with some other changes made to the same tutorial to support additional cameras. If it's still relevant to you, we'd love to hear if you have any issues with it.
Thanks!