realsenseai/librealsense #8409: How to solve the reflection problem on the floor?

laifuchicago, 2021-02-22:
To author:
I used Realsense d435i on the parking lot to detect the object. But when the reflection of the floor is detected by the realsense. Our robot will stop. So our question is-- Is there a way to avoid the reflect such as adjust the exposure or laser? Thanks.
[image]

Jonathan Sung

MartyG-RealSense, 2021-02-22:
Hi @laifuchicago   There was a past case where SLAM nevigation was being performed with a D435i, and there were repetitive patterns (objects with repeating horizontal or vertical patterns in their appearance) and a reflective floor that were generating false depth data.  This was interpreted as obstacles by their cartography program.

https://github.com/IntelRealSense/realsense-ros/issues/1343

There was another case recently with an environment similar to yours where a back wall with repetitive elements 6 meters from the camera was being observed, and the disruption on the image was reduced when a cloth was draped over the back wall to break up the pattern.

https://github.com/IntelRealSense/librealsense/issues/8258

Let's focus first on the issue of the reflections.  The negative impact on the depth image of glare from reflections can be significantly reduced by applying an external filter called a **linear polarization** filter over the lenses on the outside of the camera.  Section 4.4 of Intel's white-paper document about optical filters provides more detail about this.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

The image below, taken from that section, demonstrates the difference that the filter can make to glare reduction when applied.

[image]

If you require a scripting-based solution for sensing a reflective surface then you could try aligning depth to color, like in Intel's Python example **depth_under_water**:

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb

laifuchicago, 2021-02-23:
To author:
So how can I use the linear polarization filter in realsense viewer? Is there any tutorials or I can just tune in the realsense-viewer?

Jonathan Sung

agrunnet, 2021-02-23:
@laifuchicago sorry this is a physical filter (piece of plastic) that needs to be put on. It is not a SW feature.

laifuchicago, 2021-02-23:
To author:
So is it possible to just tune the parameters in realsense-viewer? Thanks.
Jonathan Sung

agrunnet, 2021-02-23:
No. The white paper on the Intel RealSense website should explain well how you can use a polarizer.

closed 2021-02-26 by laifuchicago