realsenseai/librealsense #8411: MATLAB Threshold Filter

piresmicael, 2021-02-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.07.100) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {  2.0 }                          | 
| Language                            |  {MATLAB }                          | 
| Segment			|  {others }                          | 

Is the threshold filter available in MATLAB?

MartyG-RealSense, 2021-02-22:
Hi @piresmicael   I do not know of a pre-made example of applying the Threshold Filter in the RealSense MATLAB wrapper.  As the MATLAB wrapper is almost 1:1 with the C++ language though, studying C++ post-processing code should be helpful guidance.

https://github.com/IntelRealSense/librealsense/issues/3311

The link below has an example of a C++ script that makes use of the threshold filter:

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

MartyG-RealSense, 2021-03-02:
Hi @piresmicael  Do you require further assistance with this case, please?   Thanks!

piresmicael, 2021-03-02:
> Hi @piresmicael Do you require further assistance with this case, please? Thanks!

No, thank you very much, I'll check it out!

closed 2021-03-02 by piresmicael