realsenseai/librealsense #8412: [L515]On the L515 does depth data deviation from color data occurs

nh-97, 2021-02-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)                                        | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C/C# }                          | 
| Segment			|  {others }                          | 

### Issue Description
I have some L515 camera problems, can anybody help me.
1. Does the depth data with the color deviation occur on L515?
2. When the data depth and color are different from the resolution, can it align? good accuracy?
Thanks for any answers!

matyhu, 2021-12-31:
I have the same question. I only used the L515 to record the color pictures, but found the white places all changed to be bule. I also want to konw the reason.

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense