realsenseai/librealsense #8413: When the D435i camera acquires a 1080p color image, the time stamp appears the same

chenxiaocongAI, 2021-02-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
const int width = 1920;
const int height = 1080;
const int fps = 30;
int delta_time=-12;
bool flag_timestamp_finish=0;
mutex mutex_delta_time;
bool start_get_image=false;
void timestamp_diff_time();
int main(int argc, char * argv[]) try
{
    rs2::pipeline pipe;
    rs2::config pipe_config;
    pipe_config.enable_stream(RS2_STREAM_COLOR , width, height, RS2_FORMAT_BGR8, fps);
    pipe_config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, fps);
    rs2::pipeline_profile profile = pipe.start(pipe_config);
//    auto depth_stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    rs2::device selected_device = profile.get_device();
    auto depth_sensor = selected_device.first<rs2::depth_sensor>();
    if (depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED))
    {
        depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f); //close emitter
    }
    std::int32_t count = 0;
    namedWindow("Test1");
    VideoWriter writer("is_same_timestamp_test.avi", CV_FOURCC('M', 'J', 'P', 'G'), 30.0, Size(width, height));
    ofstream fs_timestamp("is_same_timestamp.txt");
    struct timeval tmval;
    delta_time=0;
//    thread task02(timestamp_diff_time);
//    task02.detach();
    auto begin=std::chrono::high_resolution_clock::now();
    while(1)
    {
        // Block program until frames arrive
        rs2::frameset frameset = pipe.wait_for_frames();
        frame color_frame=frameset.first_or_default(RS2_STREAM_COLOR);
        cout<<"frameset.size() = "<<frameset.size()<<endl;
        long long color_frame_get_timestamp=color_frame.get_timestamp();
        Mat color_image(Size(width, height), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
//        cv::imshow("result",color_image);
        fs_timestamp<<color_frame_get_timestamp<<endl;
        writer <<color_image;
//        mutex_delta_time.unlock();
        count++;
        char key = static_cast<char>(cv::waitKey(1));
        if (key == 27 || key == 'q' || key == 'Q') {  // ESC/Q
            flag_timestamp_finish=1;
            break;
        }
    }
    auto end=std::chrono::high_resolution_clock::now();
    cout<<"count frame = "<<count<<endl;
    cout<<"per frame average_time = "<<double(std::chrono::duration_cast<std::chrono::milliseconds>(end - begin).count())/count<<" ms"<<endl;
    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

the save timestamp is :
1613997098541
1613997098578
1613997098617
1613997098653
1613997098689
1613997098689
1613997098762
1613997098762
...

MartyG-RealSense, 2021-02-22:
Hi @chenxiaocongAI   A RealSense team member provides an explanation for repeating RGB timestamps in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360037286634/comments/360009642013

chenxiaocongAI, 2021-02-23:
> Hi @chenxiaocongAI A RealSense team member provides an explanation for repeating RGB timestamps in the link below.
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/360037286634/comments/360009642013

Is there any way to prevent them from having duplicate timestamps .I want to synchronize the two sensors. If the camera has the same time stamp, the data will not be aligned.

chenxiaocongAI, 2021-02-23:
I want to know if there is any way to prevent RGB from dropping frames, and the timestamps are different (each frame of picture is also different)

MartyG-RealSense, 2021-02-23:
The link below features a RealSense project where its developers minimized dropped frames and duplicate timestamps by creating a custom synchronization solution in C++.  This solution is discussed in the section of the discussion headed **Custom synchronization solution**

https://medium.com/researchsummer/hand-and-body-detection-using-rgb-depth-and-thermal-camera-90b781ab4cb4

According to the article, the developers created two separate threads that are polling for frames every few milliseconds (1000/FPS) and inserted blank frames if no data arrived during the time window.

Alternatively, as you are using two streams (depth and RGB), if you wish to reduce the likelihood of dropped frames then changing the **frame queue size** from the default value of '1' to a value of '2' may help by introducing additional latency.

https://github.com/IntelRealSense/librealsense/issues/6448#issuecomment-633041074

chenxiaocongAI, 2021-02-23:
> The link below features a RealSense project where its developers minimized dropped frames and duplicate timestamps by creating a custom synchronization solution in C++. This solution is discussed in the section of the discussion headed **Custom synchronization solution**
> 
> https://medium.com/researchsummer/hand-and-body-detection-using-rgb-depth-and-thermal-camera-90b781ab4cb4
> 
> According to the article, the developers created two separate threads that are polling for frames every few milliseconds (1000/FPS) and inserted blank frames if no data arrived during the time window.
> 
> Alternatively, as you are using two streams (depth and RGB), if you wish to reduce the likelihood of dropped frames then changing the **frame queue size** from the default value of '1' to a value of '2' may help by introducing additional latency.
> 
> [#6448 (comment)](https://github.com/IntelRealSense/librealsense/issues/6448#issuecomment-633041074)
> 
> Further discussion about minimizing frame drop can be found here:
> 
> [#7663 (comment)](https://github.com/IntelRealSense/librealsense/issues/7663#issuecomment-717873757)

If you use your existing API, can’t you solve the problem of duplicate timestamps and duplicate images? If your 1080p has repeated timestamps up to 30hz, don’t mislead customers on the official website

chenxiaocongAI, 2021-02-23:
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true    can solve repeated timestamps?

chenxiaocongAI, 2021-02-23:
Are there have examples where reading RGB images will not produce the same image and time stamp

chenxiaocongAI, 2021-02-23:
the problem is solve by using cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true. Although the problem is solved, I don’t know where the problem is and how it was solved

MartyG-RealSense, 2021-02-23:
If you are building with **-DCMAKE_BUILD_TYPE=release** to build librealsense in **Release** mode instead of in **Debug** mode then librealsense can take advantage of speed-increasing optimizations.

Building with ** -DFORCE_RSUSB_BACKEND=true** also bypasses the Linux kernel when building librealsense for Linux and removes the need for patching.  This can help to resolve problems that might have been being caused by conflicts with the kernel.

chenxiaocongAI, 2021-02-24:
I found that just using -DCMAKE_BUILD_TYPE=release will cause the same time difference problem, and still need to rely on color_sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE,5) to reduce repeated frames .Will this problem be solved in the next version of the SDK or firmware ? @MartyG-RealSense

MartyG-RealSense, 2021-02-24:
Whilst the librealsense SDK is continuously improved over time with each new version, there is not a single change that can eliminate the possibility of frame drops and duplicate timestamps.

How is performance in your application affected if you disable **global time** by setting **RS2_OPTION_GLOBAL_TIME_ENABLED** to a value of '0' (disabled)?

https://github.com/IntelRealSense/librealsense/issues/5067#issuecomment-544548652

chenxiaocongAI, 2021-02-25:
If you turn off the global time, the timestamp does not seem to be repeated. Not sure if your time synchronization affects your frame rate. I want to ask what time the frame.timestamp() gets? Is this frame completed at the bottom layer .Mainly I found that there are several representative frame times, but I don’t know the meaning of each time.

MartyG-RealSense, 2021-02-25:
Further information about how global time works when enabled can be found here:

https://github.com/IntelRealSense/librealsense/pull/3909

The link below provides a good summary explanation of the different types of timestamp, including **frame_timestamp**.

https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803

I am not aware of a reduction in FPS when global time is enabled.

chenxiaocongAI, 2021-02-25:
thank you ,if i use c code to access image.RS2_OPTION_FRAMES_QUEUE_SIZE how to add  it,

int main()
{
    rs2_error* e = 0;

    // Create a context object. This object owns the handles to all connected realsense devices.
    // The returned object should be released with rs2_delete_context(...)
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    check_error(e);

    /* Get a list of all the connected devices. */
    // The returned object should be released with rs2_delete_device_list(...)
    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    check_error(e);

    int dev_count = rs2_get_device_count(device_list, &e);
    check_error(e);
    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return EXIT_FAILURE;

    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    check_error(e);

    print_device_info(dev);

    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
    rs2_pipeline* pipeline =  rs2_create_pipeline(ctx, &e);
    check_error(e);

    // Create a config instance, used to specify hardware configuration
    // The retunred object should be released with rs2_delete_config(...)
    rs2_config* config = rs2_create_config(&e);
    check_error(e);

    // Request a specific configuration
    rs2_config_enable_stream(config, STREAM, STREAM_INDEX, WIDTH, HEIGHT, FORMAT, FPS, &e);
    check_error(e);
    rs2_pipeline_profile* pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);
    check_error(e);
    // Start the pipeline streaming
    // The retunred object should be released with rs2_delete_pipeline_profile(...)
    if (e)
    {
        printf("The connected device doesn't support color streaming!\n");
        exit(EXIT_FAILURE);
    }

    rs2_device* color_sensor=rs2_pipeline_profile_get_device(pipeline_profile,&e);
    check_error(e);
//    rs2_set_option((const rs2_options*)color_sensor,RS2_OPTION_FRAMES_QUEUE_SIZE,5,&e);
//    check_error(e);
    std::int32_t count = 0;
    unsigned long long frame_number=0;
    cv::namedWindow("Test1");
    std::ofstream fs_timestamp("is_same_timestamp.txt");
    cv::VideoWriter writer("is_same_timestamp_test.avi", CV_FOURCC('M', 'J', 'P', 'G'), 30.0, cv::Size(WIDTH, HEIGHT));
    int last_number=0,current_number=0;

    auto begin=std::chrono::high_resolution_clock::now();
    while (1)
    {
        // This call waits until a new composite_frame is available
        // composite_frame holds a set of frames. It is used to prevent frame drops
        // The returned object should be released with rs2_release_frame(...)
//        cout<<"start_get_image = "<<start_get_image<<endl;
[... 53 lines cut ...]
        char key = static_cast<char>(cv::waitKey(1));
        if (key == 27 || key == 'q' || key == 'Q') {  // ESC/Q
            flag_timestamp_finish=1;
            break;
        }
    }
    auto end=std::chrono::high_resolution_clock::now();
    std::cout<<"count frame = "<<count<<std::endl;
    std::cout<<" frame number = "<<frame_number<<std::endl;
    std::cout<<"frame number - count frame  = "<<frame_number-count<<endl;
    std::cout<<"per frame average_time = "<<double(std::chrono::duration_cast<std::chrono::milliseconds>(end - begin).count())/frame_number<<" ms"<<std::endl;
    // Stop the pipeline streaming
    rs2_pipeline_stop(pipeline, &e);
    check_error(e);

    // Release resources
    rs2_delete_pipeline_profile(pipeline_profile);
    rs2_delete_config(config);
    rs2_delete_pipeline(pipeline);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}

MartyG-RealSense, 2021-02-25:
I researched your question above about how to add code changing the frame queue size to your script.  The frame buffer documentation states though that scripts that use the **pipeline** - pipe.start() - cannot set the frame queue capacity with instructions and must instead modify the source code of the SDK file **pipeline.cpp**.

Changing the frame queue size therefore may not be a practical course of action for your project as a way to imprive performance.  It is encouraging though that turning off global time has stopped duplicate timestamps.

chenxiaocongAI, 2021-02-26:
I use framebuf through a few lines of statements, and I don’t need to modify the source code.   
    rs2::device selected_device = profile.get_device();
    auto color_sensor = selected_device.first<rs2::color_sensor>();
    color_sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE,5);
Does writing like this turn on the effect?

MartyG-RealSense, 2021-02-26:
Although I have not performed this procedure myself, the formatting seems logically correct.  Like when setting exposure for the color sensor instead of the depth sensor though, my expectation would be that an index number should be used with the **profile.get_device().query_sensors()** instruction to ensure that the color sensor and not the depth sensor is having the change applied to it.  The index number is '0' for the depth sensor and '1 for the color sensor.

https://github.com/IntelRealSense/librealsense/issues/3558#issuecomment-548743478