realsenseai/librealsense #8416: Ubuntu 20 / python 3.8 support

ShettyHarapanahalli, 2021-02-22:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|844212071706|
|**Firmware Version**|05.12.11.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|904523050880|
|**Firmware Update Id**|904523050880|

Please provide a description of the problem- 
I am trying to install the python wrapper with "pip3 install pyrealsense2" but it did not work, so tried to install with build with the cmake command "cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true" I get the below error - 

/home/suman/sandbox/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory
   23 | #include "config.h"

I did install the usblib but of no use -  
sudo apt-get install libusb-1.0-0-dev

Can you please help me fix the same.

MartyG-RealSense, 2021-02-22:
Hi @ShettyHarapanahalli   If you experiencing the error **<username>/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory 23 | #include "config.h"** then advice about a fix for it is provided in the link below:

https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842

ShettyHarapanahalli, 2021-02-22:
Thanks for the link. I got a big list of warnings but installed it successfully.

closed 2021-02-22 by ShettyHarapanahalli

ShettyHarapanahalli, 2021-02-23:
Any help with this, please - 

suman@SumanDL:~/sandbox/librealsense/build$ python3
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> pipeline = rs.pipeline()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

MartyG-RealSense, 2021-02-23:
Hi @ShettyHarapanahalli   I usually see this error occur on Nvidia Jetson Nano computing boards.  Are you using a Jetson board please?

ShettyHarapanahalli, 2021-02-23:
This is on laptop.

MartyG-RealSense, 2021-02-23:
The pip package installation would not work because at the time of writing this, pip installation up to Python 3.7 only is supported (though Intel have been actively working on adding support for newer Python versions recently).

In the meantime, building the pyrealsense2 wrapper from source code with CMake like you have been doing is the best option for Python versions newer than 3.7.

Though you are using a laptop, there was a Jetson case involving **AttributeError: module 'pyrealsense2' has no attribute 'pipeline'** where the solution in that particular case was found to be to copy .so files to the same directory that the user's project script was in.

https://github.com/IntelRealSense/librealsense/issues/7078#issuecomment-672949007

This same approach of copying the .so files into the same directory as the user's project script also worked in another case involving Pi:

https://support.intelrealsense.com/hc/en-us/community/posts/360048396854/comments/360012486393

ShettyHarapanahalli, 2021-02-24:
I see only one file getting generated for some reason. Can you please help me fix this - 
 
Cmd line dump - 
build$ ls *.so
librealsense2.so

MartyG-RealSense, 2021-02-24:
Okay, first please copy **librealsense2.so** to the same folder that your project script is in.  Then find your **pyrealsense2.so** file and copy this into the folder alongside your project script too.

When using Python 3, the **pyrealsense2.so** file will not be called exactly that but have an unusual name based on the version of Python that you are using (such as **pyrealsense2.cpython-37m-arm-linux-gnueabihf.so** or something complex like that).  

I recall that when I was performing a pyrealsense2 installation, I found it by going to the parent folder of the Python installation (e.g Python37) and doing a file search from that folder location for the search term **pyrealsense2.so**.  This search located where the strangely named version of the pyrealsense2.so file was on my computer so that I could copy it into the folder where my project script was.

ShettyHarapanahalli, 2021-02-24:
I see the opengl warning in the cmake process.   
will try to follow as shared in #7969 {  https://github.com/IntelRealSense/librealsense/issues/7969 }

CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  examples/CMakeLists.txt:14 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  tools/CMakeLists.txt:28 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

ShettyHarapanahalli, 2021-02-24:
added this in 1st line of the file- librealsense/CMake/opengl_config.cmake 
set(OpenGL_GL_PREFERENCE "GLVND") 

Path of the file - 
librealsense/build//librealsense2.so 
librealsense/build/wrappers/python/pybackend2.cpython-38-x86_64-linux-gnu.so

MartyG-RealSense, 2021-02-25:
Regarding having copied **pybackend2.cpython-38-x86_64-linux-gnu.so** to beside your project script, i don't believe that this is the correct file.  It should start with 'pyrealsense2' and end in '.so' like the **pyrealsense2.cpython-37m-arm-linux-gnueabihf.so** example file-name structure that I mentioned earlier.

If an installation has numerous problems that require multiple fixes to be applied then it is usually easier to go back to the beginning, start the process over again and try to find out what the problem with the previous installation may have been.

For example, in your build you are using CMake to build librealsense and the Python wrapper together from source code, using **-DFORCE_RSUSB_BACKEND=true**.  In this build method, there is no kernel patching required.

A flag that is usually included in a CMake build statement that is building librealsense at the same time as building librealsense though is **-DPYTHON_EXECUTABLE**.  This condition defines which Python installation on the computer should be targetted.  Your CMake instruction at the start of this case does not include this flag.   **-DPYTHON_EXECUTABLE** should be set to = the exact path where your Python 3.8 installation is located on your computer.

In the pyrealsense2 installation documentation, this is Step 3.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md