realsenseai/librealsense #8419: How to verify D455 full slave mode?

daziFR, 2021-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.10.0 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            | SDK 2.0 (build 2.41.0) }                          | 
| Language                            |  cpp             | 
| Segment			|  others                           | 

### Issue Description
We tested full slave mode (Inter Cam Sync Mode set to 3)  as part of the D455 camera evaluation. Can you give us some guidance on how to verify if this is really working as expected?

Test consist of: 
 - Precise external trigger generator set to 30.0 FPS (instead of another D455 camera)
 - Inter Cam Sync Mode set to 3
 - Stereo 848x480x30
 - RGB 848x480x30
We tested both with auto-exposure and manual exposure for RGB and Stereo sensors set to the same value.

As I expected, when full slave synchronization is enabled, HW timestamps for the RGB and Stereo sensors should match (consequently FPS must match). But with the settings above, RGB HW timestamps drift over time regarding the Stereo timestamps. Also, the frame rate doesn't match : for the Stereo we achieved 30.0 when trigger is applied. For the RGB it remains 29.98Hz.

Thank You.

MartyG-RealSense, 2021-02-23:
Hi @daziFR   The advice I have from Intel is that the **External Synchronization** system is not a "mature" tool, whereas the original multiple camera system (Inter Cam Sync Mode 1 and 2) is mature and validated.  You are free to experiment with External Syncronization though.

sam598, 2021-02-23:
@MartyG-RealSense While @daziFR is using an external signal I am curious, is Inter Cam Sync Mode 3 still considered experimental if the signal is coming from a camera set to Inter Cam Sync Mode 1?

MartyG-RealSense, 2021-02-23:
Modes 3 and above are part of the **External Synchronization** system.  So if the Full Slave mode is being used then it is part of that "unvalidated" system.  The "mature" hardware sync system is the Mode 1 and 2 one represented by the original multiple camera white paper, which does not have a Full Slave mode.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

RealSenseSupport, 2021-02-23:
The RGB is not synced in mode 3, only depth is synced.

sam598, 2021-02-23:
@MartyG-RealSense and @RealSenseSupport in the whitepaper in appendix A it says:

>Setting Option value to 2 or 3 will set the Depth Camera to SLAVE or FULL SLAVE mode. The only difference between these two modes is that full slave mode also synchronizes the color camera.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-appendix-a

According to this it sounds like mode 3 is part of the "standard" sync modes. Mode 4 and above is the external sync mode.

The question is; is mode 3 "mature" and does it sync color like it says in the white paper?

RealSenseSupport, 2021-02-23:
Mode 3 is NOT mature yet. RGB sync will be implemented later.

sam598, 2021-02-23:
@RealSenseSupport can the D455 sync RGB color in Mode 2?

RealSenseSupport, 2021-02-23:
@sam598 Not yet. This feature will be implemented soon.

sam598, 2021-02-23:
So to answer @daziFR 's original question, there is no way to verify sync of the D455 because the camera itself cannot sync RGB in **ANY** mode?

@RealSenseSupport is there anything more specific than "soon"? Is it planned for the next release at the end of March? The D455 went on sale almost 6 months ago and it is still missing synchronized global shutter RGB-D, one of its key features.

RealSenseSupport, 2021-02-23:
First, the RGB and depth are synced in one cameras.
Second, the depth can be synced among multi cameras.
Yes. You are absolutely right! I will escalated this feature to dev team.

daziFR, 2021-02-24:
Thank you @RealSenseSupport, @MartyG-RealSense  for the answers and @sam598 for additional questions. Can this topic be open until there is some announcement on the firmware update related to RGB sync?

MartyG-RealSense, 2021-02-24:
Hi @daziFR   The case https://github.com/IntelRealSense/librealsense/issues/7502 is being used as the central location for information related to D455 RGB sync.

https://github.com/IntelRealSense/librealsense/issues/7502

daziFR, 2021-02-25:
In that case, I'll follow #7502. This topic can be closed. Thanks @MartyG-RealSense

MartyG-RealSense, 2021-02-25:
You are very welcome @daziFR - thanks very much

closed 2021-02-25 by MartyG-RealSense