realsenseai/librealsense #8421: Skeleton Tracking: Internal Error Occured. Application will now close. Error Details:  "Couldn't resolve requests"

meschi89, 2021-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (Open RealSense Viewer --> 2.42.0.2845) | 
| Operating System & Version |   {Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C or C++ }                          | 

### Issue Description
I have installed the IntelRealSense Viewer and everything looks as it is working.
Then I installed the Skeleton Tracking SDK, entered my trial key and wanted to start the example .exe

When I start the /bin/csharp-interactive.exe and click the play button, it shows the message "Internal Error Occured. Application will now close.
Error Details: "Couldn't resolve requests""
The application closes afterwards automatically.

When i start the "bin/cpp-realsense.exe" I get something similar:
"Exception encountered starting the RealSense pipeline: Couldn't resolve requests"

**To just want to test the SDK, this first impression with a bug is awful and represents the quality of the Skeleton Tracking SDK. Please fix.**

The bin/csharp-image.exe returns the following output:

Skeleton tracking will be run on an image saved at %LOCALAPPDATA%\res\images\skeleton_estimation.jpg...
[CUBEMOS_LOG_1][2021-Feb-23 11:59:24.600198][info]Cubemos Logging initialised
[CUBEMOS_LOG_2][2021-Feb-23 11:59:24.610233][info]Initialising the cubemos skeleton tracking v3.1.0
[CUBEMOS_LOG_3][2021-Feb-23 11:59:34.187684][warning]No hard disk drives could be found. The activation serial key would not use an HDD signature.
[CUBEMOS_LOG_4][2021-Feb-23 11:59:34.188034][info]The provided activation key has been accepted for creation of the cubemos handle.
[CUBEMOS_LOG_5][2021-Feb-23 11:59:34.215277][info]Loaded the cubemos plugin
[CUBEMOS_LOG_6][2021-Feb-23 11:59:37.733847][info]Provided model loaded to the cubemos plugin.
# Persons detected: 1
Skeleton #0
        Joint coordinate #0: 1572.36352539063; 1506.796875
        Joint coordinate #1: 1639.27270507813; 1841.640625
        Joint coordinate #2: 1371.63635253906; 1841.640625
        Joint coordinate #3: 1304.72717285156; 2243.453125
        Joint coordinate #4: 1438.54541015625; 2645.265625
        Joint coordinate #5: 1839.99987792969; 1841.640625
        Joint coordinate #6: 1906.90905761719; 2243.453125
        Joint coordinate #7: 1706.18176269531; 2578.296875
        Joint coordinate #8: 1505.45446777344; 2712.234375
        Joint coordinate #9: 1572.36352539063; 3381.921875
        Joint coordinate #10: 1572.36352539063; 3917.671875
        Joint coordinate #11: 1773.0908203125; 2712.234375
        Joint coordinate #12: 1906.90905761719; 3314.953125
        Joint coordinate #13: 1973.81811523438; 3917.671875
        Joint coordinate #14: 1505.45446777344; 1439.828125
        Joint coordinate #15: 1639.27270507813; 1439.828125
        Joint coordinate #16: 1438.54541015625; 1506.796875
        Joint coordinate #17: 1706.18176269531; 1506.796875
Press <Enter> to exit...

MartyG-RealSense, 2021-02-23:
Hi @meschi89   My understanding from your description is that the **Cubemos Skeleton Tracking SDK** example program **csharp-interactive.exe** did not work on your computer but the **bin/csharp-image.exe** did work for you.  Is that correct please?

The Cubemos Skeleton Tracking SDK is not an Intel product and technical support for it is provided by the **Cubemos** company.  They can be contacted at the link below.

https://www.cubemos.com/skeleton-tracking-sdk

The Cubemos SDK is a separate software from the RealSense SDK and its RealSense Viewer program, which Intel does provide technical support for.

meschi89, 2021-02-23:
Hello MartyG

The csharp-image.exe results a skeleton with 18joints within the image "skeleton_estimation.jpg" (the example image) -> correct

the csharp-interactive shows (see screenshot, stops after dialog approval) -> not good
[image]

The csharp-realsense shows
Initializing console Skeleton Tracking sample with RealSense ...
Hold the Intel RealSense with person(s) in scene and hit <ENTER>...

Unhandled Exception: System.Exception: Couldn't resolve requests ---> System.Runtime.InteropServices.ExternalException: rs2_pipeline_start_with_config(pipe:0000024ACE9B5450, config:0000024ACE9B5770)

   --- End of inner exception stack trace ---
   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData)
   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
   at Intel.RealSense.NativeMethods.rs2_pipeline_start_with_config(IntPtr pipe, IntPtr config, Object& error)
   at Intel.RealSense.Pipeline.Start(Config cfg)
   at Cubemos.Samples.Program.Main(String[] args)

I hope that this gives some insights

MartyG-RealSense, 2021-02-23:
Intel does not provide technical support for the Cubemos SDK so we cannot offer advice about your problem with it.  Please provide your image and results to Cubemos.   Thanks!

meschi89, 2021-02-23:
Ok, I will contact them. Thanks for your help.
The Intel RealSense Viewer 2.42.0 works as expected.

However, in comparison to the 'Azure Kinect' the 'RealSense D435' stands no chance.
Thanks for your help.

MartyG-RealSense, 2021-03-02:
Case closed due to correct answer provided.

closed 2021-03-02 by MartyG-RealSense