realsenseai/librealsense #8423: pipe.start(cfg) with net_device

controller1304, 2021-02-23:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Operating System & Version | Win (10)                                                  | 
| Platform                                 | PC                                                           |
| SDK Version                           |  2.41.0                                                     | 
| Language                               |  C++                                                       | 
| Segment		         	|  Robot                                                     | 

I am trying to adapt the rs_measure example, so that it connects to a camera, that is connected to a raspberry pi.

I change 

rs2::pipeline pipe

 to

rs2::net_device dev("10.0.0.99");
rs2::context ctx; dev.add_to(ctx);
rs2::pipeline pipe(ctx);
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

pipe.start(cfg) does not work. It says "No connection to device".
pipe.start() does work

cfg.can_resolve(pipe); says always  false

Is it possible to start a net_device with a configuration?

MartyG-RealSense, 2021-02-23:
Hi @controller1304   In your full script, have you placed the line below directly before your first line that uses cfg so that the SDK knows what **cfg:** refers to?

**rs2::config cfg;**

For example:

**rs2::config cfg;
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));**

controller1304, 2021-02-23:
Yes that is what I did:
        rs2::net_device dev("192.168.0.194");
	rs2::context ctx; dev.add_to(ctx);
	rs2::pipeline pipe(ctx);
	rs2::config cfg;
	cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
	cfg.enable_stream(RS2_STREAM_DEPTH);
	cfg.enable_stream(RS2_STREAM_COLOR,640,480, RS2_FORMAT_RGBA8);// Enable default depth
	bool ressolve=cfg.can_resolve(pipe);
	cout<<ressolve;
	cout<<"\n";
	pipe.start(cfg);

output:

0
RealSense error calling rs2_pipeline_start_with_config(pipe:0000000005D2A0E0, config:0000000005D2AE00):
    No device connected

MartyG-RealSense, 2021-02-23:
Is it a multiple camera setup or are you using a single camera, please?

controller1304, 2021-02-23:
There is one camera D435i connected to a raspberry pi, which sits on a robot.
On the raspberry pi there runs the image that is published here:
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras
On the PC side I want to run the same program that I developed, with the camera directly connected to the PC. It is an adaption of rs-measure example which needs more computing power then the raspi has. I want to detect aruco markers, which are mounted on the robot arm to move the arm to the correct position.

MartyG-RealSense, 2021-02-23:
When you test with **pipe.start()** without cfg in the brackets, it is likely working because the pipeline is not accessing the cfg configuration details in the preceding lines and is using the camera's default configuration instead.  Without cfg in the brackets, the pipeline does not know that the custom configuration exists.  So when cfg is included in the brackets so that the SDK accesses the configuration details, this suggests that there is something in the configuration that the project is failing on.

My feeling is that the cfg lines used here may be more suited to an application that is accessing multiple cameras from the same program (like **rs-multicam**) and that it could use a simpler design if it only has to access one camera, and not use a serial number (which is helpful for distinguishing between multiple attached cameras).  For example:

**rs2::pipeline pipe;
//Create a configuration for configuring the pipeline with a non default profile
rs2::config cfg;
//Add desired streams to configuration
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_stream(RS2_STREAM_COLOR,640,480, RS2_FORMAT_RGBA8);
//Instruct pipeline to start streaming with the requested configuration
pipe.start(cfg);**

controller1304, 2021-02-23:
The recommended code snippet works only if I connect the camera directly to the PC. But as soon as I connect the camera to the raspberry pi and run the recommended code on the PC :

**rs2::net_device dev("192.168.0.194");
rs2::context ctx; dev.add_to(ctx);
rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH);
cfg.enable_stream(RS2_STREAM_COLOR,640,480, RS2_FORMAT_RGBA8);
pipe.start(cfg);**

I get the output:

_RealSense error calling rs2_pipeline_start_with_config(pipe:0000000006DF1200, config:0000000006DF1250):
    No device connected_

MartyG-RealSense, 2021-02-24:
I located an rs2::net case from May 2020 that had the same error with using the serial number over the network.

https://github.com/IntelRealSense/librealsense/issues/6376

The RealSense user in that case posted a workaround script that had the limitation that it could only access the last camera detected.  As you only have one camera in your project though, this script may provide a solution.

controller1304, 2021-02-25:
In the solution in #6376 it also ended that 
**pipe.start(cfg)**
was changed to
**pipe.start()**
what I do not want, because the then the frames arrive in the wrong format, and I have no possibility to reduce the framerate.

MartyG-RealSense, 2021-02-25:
I mentioned this case earlier today to the developer who authored rs2-net and they are aware of the case and your issue with cfg.  I do not have a solution to suggest at this time though other to use pipe.start(), even though the default color stream setting has a higher resolution.

It is worth mentioning that the cfg instructions that you are using would not directly define frames per second, as they define resolution but not FPS.  FPS is defined in cfg statements by adding a number at the end of the bracket representing an FPS speed supported by your camera model.  for example, to set 640x480 color at **30 FPS**, the cfg definition would be:

cfg.enable_stream(RS2_STREAM_COLOR,640,480, RS2_FORMAT_RGBA8**, 30**);

controller1304, 2021-03-01:
No, there is no difference.

MartyG-RealSense, 2021-03-02:
I have re-examined this case from the beginning and thoroughly checked the code against other references but cannot find anything wrong with it.

Have you tested whether removing the can_resolve check (which checks whether the camera can satisfy the defined configuration) makes a difference?

[image]

MartyG-RealSense, 2021-03-08:
Hi @controller1304  Do you require further assistance with this case, please?  Thanks!

controller1304, 2021-03-08:
Yes I need further assistance and would like to know how to solve this issue. No, that makes no difference, if I remove the ressolve command:

**rs2::net_device dev("192.168.0.194");
rs2::context ctx; dev.add_to(ctx);
rs2::pipeline pipe(ctx);
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGBA8, 30);
auto profile = pipe.start(cfg);
cout<<"started";**

Output:

_RealSense error calling rs2_pipeline_start_with_config(pipe:0000000006D573C0, config:0000000006D576E0):
    No device connected_

MartyG-RealSense, 2021-03-09:
I investigated your case from the beginning again to seek new insights.  I recalled that the current version of **rs-server** only supports a limited number of resolution modes, and RGB8 / RGBA8 is not among them,  This may be why cfg is not working, because a compatible profile cannot be found for the **network camera**.  For a 30 FPS color stream from the RGB sensor, rs-server supports **YUY2** at 640x480.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

Does it work if you change the color format RGBA8 to **RS2_FORMAT_YUY2** please?

MartyG-RealSense, 2021-03-10:
Thanks very much for the update @controller1304 - I'm very pleased to hear that it is working for you!

controller1304, 2021-03-22:
Thanks for yout help, here you see the outcome:
https://www.youtube.com/watch?v=nlNbJYR-Emw

MartyG-RealSense, 2021-03-22:
Great video @controller1304 - thanks so much for sharing it   :)