realsenseai/librealsense #8425: Issue building nodejs wrapper

holoxica-cb, 2021-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.10.00 | 
| Operating System & Version |   Win 10  |  
| Platform                                 | PC  |
| SDK Version                            |  2.42.0                          | 
| Language                            |  nodejs                         | 
| Segment			|  Other                          | 

### Issue Description
I am attempting to build the nodejs bindings so that I can use the camera in a nodejs (nwjs) project. Some other issues suggest that the node-librealsense npm module is [outdated](https://github.com/IntelRealSense/librealsense/issues/4242#issuecomment-503528583) so I am building from source.

I have been following the instructions [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs). I am new to CMake but I think I am following the right steps, configuring then generating the files with `-DBUILD_NODEJS_BINDINGS=1`.
[image]

I get to step 1.3 but here I hit a problem. Using the "Build with CMake" approach in the command line or using the GUI above, the `make` command always gives me `make: *** No targets specified and no makefile found.  Stop.`, with me attempting this in the main, \build and \wrappers\nodejs directories. I'm assuming this step actually makes the .node file required (not present currently).

If I try the "Manually Build" approach, I get the following error regarding missing files when running `npm install`: 
[image]

MartyG-RealSense, 2021-02-23:
Hi @holoxica-cb   There was a lengthy case with similar NodeJS installation problems at the link below.

https://github.com/IntelRealSense/librealsense/issues/7439

During the course of that investigation, there was another NodeJS case where a RealSense user developed a solution involving **npm install** that worked for them and shared it.

https://github.com/IntelRealSense/librealsense/issues/7439#issuecomment-708503008

After those two cases in September 2020, Intel published some changes to NodeJS support in SDK **2.40.0** in November 2020.

[image]

MartyG-RealSense, 2021-03-02:
Hi @holoxica-cb  Do you require further assistance with this case, please?   Thanks!

holoxica-cb, 2021-03-05:
Hi @MartyG-RealSense, looking into that solution I would've had to revert back several versions of NodeJS which would break other dependencies. I have successfully built the Python wrapper however and am giving that a try.

As previously mentioned this build process is all new to me - I didn't realise at the time that I needed use Visual Studio to run the build after generating the build files with CMake. I mention this in case anyone else is in a similar position and finds this.

closed 2021-03-05 by holoxica-cb