realsenseai/librealsense #8427: Issue with DWS SDK (Dimensional Weight Software) and Errors with enum and unresolved external symbol

thomasdriss, 2021-02-23:
Hello everyone,

I use the L515 camera and
I have an issue with the DWS SDK (Dimensional Weight Software), when I try to launch the example's code I have a lot of errors : 

[image]

- I put in the additional include directories the 3 SDK headers 
- I put in the additional Libarires directory the DWSSDK.lib 
- I put in the same folder of my .sln file all .dll files the .lib file the plugins.xml file a folder named "include" with the 3 .h files and the configuration file 

[image]

Can someone has already had the same issues ? 

Thanks a lot

PS : this is the code I entered in the sln file 

```/*******************************************************************************

INTEL CORPORATION PROPRIETARY INFORMATION
This software is supplied under the terms of a license agreement or nondisclosure
agreement with Intel Corporation and may not be copied or disclosed except in
accordance with the terms of that agreement
Copyright(c) 2017 Intel Corporation. All Rights Reserved.

*******************************************************************************/

#include "DWS_sdk.h"
#include <iostream>
#include <mutex>

void* FrameBuffers[3]; // 0 - depth, 1 - IR, 2 - RGB
void* local_depth = new unsigned char[640 * 480 * 2];
void* local_ir = new unsigned char[640 * 480 * 1];
void* local_colour = new unsigned char[1280 * 720 * 3];
bool keep_alive = true;
bool image_ready = false;
std::mutex mutex_display;
DWS_Box boxes[DWS_MAX_NUM_OF_BOXES];
int numOfDetectedBoxes = 0;
long im_id = -1;
int handler = -1;

void* _callback(const void* depth_buffer, int d_w, int d_h, int d_bpp, const void* ir_buffer, int i_w, int i_h, int i_bpp, const void* colour_buffer, int c_w, int c_h, int c_bpp, int frameId)
{
	// ###############################
	// #### CRITICAL ####
	// COPY FRAMES TO LOCAL BUFFER!!!!
	// ###############################
	{
		std::lock_guard<std::mutex> lck(mutex_display);
		memcpy(local_depth, depth_buffer, (d_w * d_h * d_bpp));
		memcpy(local_ir, ir_buffer, (i_w * i_h * i_bpp));
		if (colour_buffer != nullptr)
			memcpy(local_colour, colour_buffer, (c_w * c_h * c_bpp));
		im_id = frameId;
		image_ready = true;
	}

	return nullptr;
}

void* _dim_callback(int frameId, int box_count, const DWS_Box* box_buffer)
{
	numOfDetectedBoxes = box_count;
	return nullptr;
}

void* _status_callback(DWS_Severity severity, DWS_MessageCode code, const char* msg)
{
	std::cout << "Severity: " << severity << ", Code: " << code << ", " << msg << std::endl;

	switch (code) {
	case CALIBRATION_FAILED:	// Couldn't complete base surface detection
	case BAD_USER_INPUT:		// Bad input to dws_create
	case CAMERA_DISCONNECTED:	// Detected camera disconnection
	case INVALID_LICENSE:		// License file missing, expired, or doesn't match camera SN
	case MISSING_CONFIGS:		// Couldn't find the /Configuration/ folder
	case BAD_API_CALL:			// API handle called during an active earlier one
	case UNKNOWN_ERROR:			// Internal, exception, msg might contain more details
		dws_stop(handler);
		keep_alive = false;
	default:
		break;
	}

	if (code == CALIBRATION_SUCCESSFUL)
	{
		if (dws_start(handler, _dim_callback) != DWS_Status::SUCCESS)
			dws_stop(handler);
	}
	return nullptr;
}

int main(int argc, char* argv[])
{
	int imageNum = 0;
	auto status = dws_create(DWS_Mode::STATIC_MODE, false, _callback, _status_callback, &boxes[0], handler);
	if (status != DWS_Status::SUCCESS)
	{
		return -1;
	}
	std::string _ver = dws_version(handler);
	dws_autoCalibrate(handler, false);
	while (keep_alive) {
		if (image_ready) {
			std::lock_guard<std::mutex> lck(mutex_display);
			std::cout << "new image received. ID = " << im_id << std::endl;
			image_ready = false;
		}
	}

	return EXIT_SUCCESS;
}

RealSenseSupport, 2021-02-25:
Hi @thomasdriss 

We'll have a look.  Have you been able to run the executable that came with the package correctly?  Just to make sure we rule out any activation issues or any other issues with the registration process.

thomasdriss, 2021-02-25:
Hello @RealSenseSupport 

Yes the exe works well, do I have to put the file "Licence" in the SDK folder also ?

Thanks

thomasdriss, 2021-02-26:
Problem solved thanks

closed 2021-02-26 by thomasdriss

jnapoles, 2024-06-07:
Hi Tomas, can you help me?, I try to follow steps for my DMS sample but can't I find any files and dll 

https://dev.intelrealsense.com/docs/dimensional-weight-software-dws

> Hello everyone,
>
> I use the L515 camera and I have an issue with the DWS SDK (Dimensional Weight Software), when I try to launch the example's code I have a lot of errors :
>
> [image]
>
> * I put in the [...]

thomasdriss, 2024-06-09:
Hello jnapoles, unfortunately I am not working anymore with this product.
I wish you a very pleasant day.