realsenseai/librealsense #8433: Whether is Operating temperature differs with Projector temperature?

vishnukumarts, 2021-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          |            | 

### Issue Description
My room temperature is around 25 to 27 degree Celcius.  But my projector and ASIC temperature is 35 & 39 degree respectively.   As per the data sheet it operating temperature is 5-35 degree.  Whether do you have any data relating the operating temperature and projector/ASIC temperature.

MartyG-RealSense, 2021-02-24:
Hi @vishnukumarts   There is a relationship between the outer environmental temperature and the inner temperature of the camera.  Intel lab tests found that the projector temperature should stay below 60 degrees C if the camera skin temperature is below 44 degrees C.

https://github.com/IntelRealSense/librealsense/issues/5209#issuecomment-563358823

The camera firmware has a laser safety mechanism that checks that the projector temperature is not less than 0 degrees C or greater than 60 degrees C.  If the temperature is detected to be above 60 for more than several seconds then it first halves laser power to try to bring the temperature down and then shuts the laser off if the projector temperature continues to exceed 60. 
 The laser will also shut off if the projector temperature is less than zero degrees C.

Although your camera temperatures are high, under real-world conditions the camera should still be able to operate correctly in that temperature range.  It may become a problem if the temperatures start climbing into the 40's.  You could consider creating a custom housing for the camera that provides additional cooling in a hot environment.

vishnukumarts, 2021-02-24:
thank you.  will provide additional cooling to the camera and confirm the behaviour.

MartyG-RealSense, 2021-03-03:
Hi @vishnukumarts  Do you have an update that you can provide about your case please?  Thanks!

closed 2021-03-08 by MartyG-RealSense