realsenseai/librealsense #8435: SLAM mapping without ROS?

rxr5160, 2021-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / T265 | 
| Firmware Version                   | D435: 05.12.10.000 / T265: 0.2.0.951 | 
| Operating System & Version |   Ubuntu 18 | 
| Kernel Version (Linux Only)    |  5.4.0-65-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.4.1                           | 
| Language                            |  C/C++                           | 
| Segment			|  Robot                          | 

I am part of a group of students tasked with trying to utilize SLAM and the D435 and T265 cameras to navigate a robot through hallways and various rooms via labels on a map. In our investigations we found an issue with the T265 and a raspberry pi4 where the camera is unable to communicate with the device, a known issue and confirmed to not be addressed by the RealSense team ( https://github.com/IntelRealSense/librealsense/issues/6362#issuecomment-697971303 ). This led us to change to the Ubuntu PC where the T265 could communicate properly. However, we found the same issue to come up when trying to leverage ROS. This has caused many issues as most examples and explanations or libraries to help with SLAM and mapping use ROS and do not give much or any help outside of it. 

Recently we have started to try and leverage MIT Kimera. However, it is not very well documented in how to connect live cameras and we have struggled with learning how to do so. 

Therefore, I ask here if others have created maps and tried to navigate them using these cameras without ROS and is able to give us some pointers in how to proceed.

RealSenseSupport, 2021-03-15:
@rxr5160 Do you still have question? As this question is out of librealsense scope, I'll leave it to be answered by the community user. Or Maybe you can try to get more help from SLAM projects. Thanks a lot!

RealSenseSupport, 2021-04-06:
@rxr5160 Any update from your side? Do you still need further support on this? Thanks!

rxr5160, 2021-04-06:
No I do not. We changed the project direction. thank you.

RealSenseSupport, 2021-04-07:
@rxr5160 Thanks for the update! Close the ticket accordingly.

closed 2021-04-07 by RealSenseSupport

shaswat0102, 2022-04-11:
> Required Info	
> Camera Model	D435 / T265
> Firmware Version	D435: 05.12.10.000 / T265: 0.2.0.951
> Operating System & Version	Ubuntu 18
> Kernel Version (Linux Only)	5.4.0-65-generic
> Platform	PC
> SDK Version	2.4.1
> Language	C/C++
> Segment	Robot
> I am part of a group of students tasked with trying to utilize SLAM and the D435 and T265 cameras to navigate a robot through hallways and various rooms via labels on a map. In our investigations we found an issue with the T265 and a raspberry pi4 where the camera is unable to communicate with the device, a known issue and confirmed to not be addressed by the RealSense team ( [#6362 (comment)](https://github.com/IntelRealSense/librealsense/issues/6362#issuecomment-697971303) ). This led us to change to the Ubuntu PC where the T265 could communicate properly. However, we found the same issue to come up when trying to leverage ROS. This has caused many issues as most examples and explanations or libraries to help with SLAM and mapping use ROS and do not give much or any help outside of it.
> 
> Recently we have started to try and leverage MIT Kimera. However, it is not very well documented in how to connect live cameras and we have struggled with learning how to do so.
> 
> Therefore, I ask here if others have created maps and tried to navigate them using these cameras without ROS and is able to give us some pointers in how to proceed.

Can I know what did you applied for gmapping without implementing ROS or the course of direction hat you picked for visual odemtry without ROS?