realsenseai/librealsense #8436: Imgui on frame image.

arothenberg, 2021-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
This is probably more a programming issue then realsense. Sorry about that but I cant seem to find answers anywhere else.

I render my frame images using an int */array of rgba values. And then I custom draw circles and lines using the pointer data and   memset. Then I render it. That works fine. But I'd rather us Imgui so
1. I'm using less cpu
2. I can use gslblock code that you have. And I'm not sure that  I can do  that with my custom array. 

So, I want to switch from custom drawing my circles onto the array before rendering to using opengl circles after I render the frame using you colorizer and uploader. The problem is I don't know how to translate from opengl/imgui coordinates to the coordinates of the depth table.
In my case the depth table is 640,480 and the app window is 1280,800. 
So, to recap I want to dynamically placed a circle at certain points(say the closest point to the camera) using opengl/imgui.  

Thanks in advance. 

.

MartyG-RealSense, 2021-02-24:
Hi @arothenberg   The discussion in the link below about imgui drawing commands in librealsense may be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360052500553-How-to-draw-shapes-on-a-window-using-ImGiu-

MartyG-RealSense, 2021-03-07:
Hi @arothenberg Can you let me know if you require further assistance with this case, please? Thanks!

closed 2021-03-13 by MartyG-RealSense