realsenseai/librealsense #8438: pybackend2: get_xu-->"The property set specified does not exist"; set_pu-->HResult 0x8007001f: "A device attached to the system is not functioning."

skaudrey, 2021-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D455                | 
| Firmware Version                    | 05.12.10.00       | 
| Operating System & Version  |  Win 10             | 
| Platform                                  | PC                     |
| SDK Version                            | V 2.41.0             | 
| Language                                |  python             | 
| Segment			          | others               | 

Hi, I use D455 and try to run the example offered [here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pybackend_example_1_general.py). 

I compile my pybackend2 according to the guides, but I install pyrealsense2 ( 2.42.0.2924) by pip, then after compilation, I copied the compiled pyrealsense2.pyd and pybackend2.pyd under /Lib/site-packages.

Then when I run the script to line 53
`ae = dev.get_xu(xu, 0xB, 1) # get XU value. args: XU, control #, # of bytes`
I get the error: 
**hr returned: HResult 0x80070492: "The property set specified does not exist on the object."**

If I skip the XU controls part and go to PU controls, when it is executed to line 64
` dev.set_pu(rs.option.gain, 32)`, 
I get the error:
**hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."**

MartyG-RealSense, 2021-02-25:
Hi @skaudrey  Which version of Python are you using, please?

skaudrey, 2021-02-27:
> Hi @skaudrey Which version of Python are you using, please?

py3.6.12

MartyG-RealSense, 2021-02-27:
Could you describe your overall installation process for librealsense, Python and the Python wrapper to me, please?  It sounds as though you are using a laptop or desktop PC and Python 3.6.  If you are using a laptop / desktop PC and not a device with an Arm processor such as Raspberry Pi or Nvida Jetson then you should be able to install the wrapper using the pyrealsense2 pip package.  This would mean that you could ignore the installation instructions about building the wrapper from source code.

Below is the pip-based installation method that I have used on my own Windows computer:

1.  Install the full RealSense SDK using the Windows 10 automated installer from the SDK **Releases** page.

[image]

2.  Install Python on Windows if it is not already installed.  At the tine of writing this, if you intend to install the RealSense Python wrapper with pip then do not install a Python version newer than Python 3.7.  The guide in the link below should be helpful for installing Python on Windows.

https://www.liquidweb.com/kb/how-to-install-python-on-windows/

3.  This next stage is likely unnecessary for you if you are already usng pip but I will list it anyway for the benefit of other readers.  Install pip by downloading the file **get-pip.py** and opening a Windows command prompt at the location of the get-pip.py file.  Then input the following command to install pip: **python get-pip.py**

You can obtain the file by right-clicking on the link below and selecting the option for downloading a link.  It might be called Save Link As, Save Target As or something similar.

https://bootstrap.pypa.io/get-pip.py

This link is sourced from the **Liquid Web** guide to installing pip on Windows, which was the guide that I followed when documenting my own installation procedure for setting up pip.

https://www.liquidweb.com/kb/install-pip-windows/

4.  Place the file in the root folder of your Python installation on Windows (e.g **Python36** for Python 3.6 or **Python37** for Python 3.7).

5.  In the Windows command prompt, use the **cd** command to change directory to the **Scripts** sub-folder of the Python parent (e.g **Python37/Scripts**).  Enter the command **pip** into the Windows command prompt to check whether pip has been installed okay.

6. From that **/Scripts** folder, input the pip install command into the Windows command prompt to install the pyrealsense2 wrapper with pip:

**pip install pyrealsense2**

skaudrey, 2021-03-03:
> Could you describe your overall installation process for librealsense, Python and the Python wrapper to me, please? It sounds as though you are using a laptop or desktop PC and Python 3.6. If you are using a laptop / desktop PC and not a device with an Arm processor such as Raspberry Pi or Nvida Jetson then you should be able to install the wrapper using the pyrealsense2 pip package. This would mean that you could ignore the installation instructions about building the wrapper from source code.
> 
> Below is the pip-based installation method that I have used on my own Windows computer:
> 
> 1. Install the full RealSense SDK using the Windows 10 automated installer from the SDK **Releases** page.
> 
> [image]
> 
> 1. Install Python on Windows if it is not already installed.  At the tine of writing this, if you intend to install the RealSense Python wrapper with pip then do not install a Python version newer than Python 3.7.  The guide in the link below should be helpful for installing Python on Windows.
> 
> https://www.liquidweb.com/kb/how-to-install-python-on-windows/
> 
> 1. This next stage is likely unnecessary for you if you are already usng pip but I will list it anyway for the benefit of other readers.  Install pip by downloading the file **get-pip.py** and opening a Windows command prompt at the location of the get-pip.py file.
> 
> You can obtain the file by right-clicking on the link below and selecting the option for downloading a link. It might be called Save Link As, Save Target As or something similar.
> 
> https://bootstrap.pypa.io/get-pip.py
> 
> This link is sourced from the **Liquid Web** guide to installing pip on Windows, which was the guide that I followed when documenting my own installation procedure for setting up pip.
> 
> https://www.liquidweb.com/kb/install-pip-windows/
> 
> Place the file in the root folder of your Python installation on Windows (e.g **Python36** for Python 3.6 or **Python37** for Python 3.7).
> 
> 1. Once the **get-pip.py** file is in the root folder of your Python installation, use the **cd** command in the Windows command prompt to change directory to the Python root folder and input the following command to install pip:
> 
> **python get-pip.py**
> 
> 1. In the Windows command prompt, **cd** to the **Scripts** sub-folder of the Python parent (e.g **Python37/Scripts**).  Enter the command **pip** into the Windows command prompt to check whether pip has been installed okay.
> 2. Finally, from that **/Scripts** folder, input the pip install command into the Windows command prompt to install the pyrealsense2 wrapper with pip:
> 
> **pip install pyrealsense2**
> 
> This command should install the latest version of the pip pyrealsense2 file from the PyPi pyrealsense2 file download archive for the Python 3 version that you have got installed.

Hi, here's what I did while installing on my laptop (win10 x64):

1. My python (3.6.12) is installed by anaconda previously.
2. Then I install realsense SDK 2.0 from the link you mentioned in your reply.
3. Then run `pip install pyrealsense2`.
4. Then I try to run pybackend_example_1_general.py. But the **pybackend2 cannot be found**. So I compile the source codes according to the instructions, and _just copy the compiled realsense2.pyd, pybackend2.pyd to Lib/site-packages_. After this `import pybackend2` works.
5. Then try to run the  pybackend_example_1_general.py, I got the error I described.

B.T.W.,

I also tried uninstall all (realsense SDK and pyrealsense2), then reinstall realsense SDK, `pip install pyrealsense2`. And to make the scripts works on `import pybackend2`, directly copy files _realsense2.pyd, pybackend2.pyd_ **created by running pip** to Lib/site-packages. The error is the same.

MartyG-RealSense, 2021-03-03:
I recall from a recent case that if Conda is being used then the **PYTHONPATH** used to direct the Python wrapper to where Python is installed on the computer may be different from the ones typically used with Python.

https://github.com/IntelRealSense/librealsense/issues/8184#issuecomment-770365205

skaudrey, 2021-03-03:
> I recall from a recent case that if Conda is being used then the **PYTHONPATH** variable used to direct the Python wrapper to where Python is installed on the computer may be different from the paths typically used with Python.
> 
> [#8184 (comment)](https://github.com/IntelRealSense/librealsense/issues/8184#issuecomment-770365205)

Well, pip3 and add PYTHONPATH to system environments still don't work.

MartyG-RealSense, 2021-03-04:
I went back to the start of the case and re-read it to seek new insights.  

How I read it is that you installed the Python wrapper using **pip install pyrealsense2** but the **pybackend_example_1_general.py** example script that you tried with it did not work.

So you tried building the Python wrapper from source code instead.   You then **copy the compiled realsense2.pyd, pybackend2.pyd to Lib/site-packages**.  It seems as though this is the stage of the process where it changes from the recommended instructions and goes wrong.

The Windows version of the wrapper's instructions for building from source code ask that the **pyrealsense2.pyd** file be coped to **\Lib\site-packages** but your method copies **realsense2.pyd and pybackend2.pyd** to that folder instead.  This would suggest that your installation process should be to put  **pyrealsense2.pyd** into the Lib/site-packages folder.

In your **BTW** you then try copying **realsense2.pyd and pybackend2.pyd** into the Lib/site-packages folder.   This too does not meet the recommendation of the instructions to install **pyrealsense2.pyd** into that folder (not pybackend2.pyd).

I hope that the above information is helpful.

skaudrey, 2021-03-05:
> I went back to the start of the case and re-read it to seek new insights.
> 
> How I read it is that you installed the Python wrapper using **pip install pyrealsense2** but the **pybackend_example_1_general.py** example script that you tried with it did not work.
> 
> So you tried building the Python wrapper from source code instead. You then **copy the compiled realsense2.pyd, pybackend2.pyd to Lib/site-packages**. It seems as though this is the stage of the process where it changes from the recommended instructions and goes wrong.
> 
> The Windows version of the wrapper's instructions for building from source code ask that the **pyrealsense2.pyd** file be coped to **\Lib\site-packages** but your method copies **realsense2.pyd and pybackend2.pyd** to that folder instead. This would suggest that your installation process should be to put **pyrealsense2.pyd** into the Lib/site-packages folder.
> 
> In your **BTW** note, you then try copying **realsense2.pyd and pybackend2.pyd** into the Lib/site-packages folder. This too does not meet the recommendation of the instructions to install **pyrealsense2.pyd** into that folder (not pybackend2.pyd).
> 
> I hope that the above information is helpful.

Sorry, my bad for the **typos**. 

It's **pyrealsense2.pyd**, always.

Here I list what I have under Lib/site-packages

[image]

MartyG-RealSense, 2021-03-05:
Thanks for the update.  The filename and location of your pyrealsense.2 pyd file look okay.  And you have already defined your PYTHONPATH variable.

Which path to your Python Conda did you use for PYTHONPATH, please?

*********************

If placing pyrealsense2.pyd into the Lib/site-packages folder is not working for you then you still have the alternate method available of placing files in the same folder as your Python test script by copying **pyrealsense2.cp36-win_amd64.pyd and realsense2.dll** to that folder.

You can find pre-made x64 versions of these two files for the particular Windows librealsense SDK version that you are using by navigating to the SDK folder location below:

 **C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64**
 
[image]

skaudrey, 2021-03-06:
> Thanks for the update. The filename and location of your pyrealsense.2 pyd file look okay. And you have already defined your PYTHONPATH variable.
>
> Which path to your Python Conda did you use for [...]

Hi, thank you for your patience.

Here's the setting of environment variables:
[image]
where 1 is the path of my anaconda, 2 is the _PYTHONPATH_ and _Path_ variable, 3 is the concrete value of _Path_. 

P.S., Under this envorionment setting, try to copy the files you mentioned to the path where the script is doesn't help.

MartyG-RealSense, 2021-03-06:
I located a complete guide for doing a Python wrapper installation on Anaconda and Windows that a RealSense user created based on advice provided earlier in the same case by a RealSense team member.

https://github.com/IntelRealSense/librealsense/issues/1657#issuecomment-388734684

skaudrey, 2021-03-09:
> I located a complete guide for doing a Python wrapper installation on Anaconda and Windows that a RealSense user created based on advice provided earlier in the same case by a RealSense team member and another RealSense user.
> 
> [#1657 (comment)](https://github.com/IntelRealSense/librealsense/issues/1657#issuecomment-388734684)
> 
> As well as reading this guide, it is worth reading the case from the beginning too to get a full understanding of what is being discussed.

Sorry, follow the guide completely, still no progress.

MartyG-RealSense, 2021-03-10:
There is limited information available about installing the Python wrapper on Windows and Conda, so I do apologize for the lack of progress.

There is an alternative Windows / Conda installation guide at the link below that may provide some new insights.

https://rahulvishwakarma.wordpress.com/2019/08/17/realsense-435i-depth-rgb-multi-camera-setup-and-opencv-python-wrapper-intel-realsense-sdk-2-0-compiled-from-source-on-win10/

closed 2021-03-23 by MartyG-RealSense