realsenseai/librealsense #8447: Question on L515 Multi-camera sync video shooting

seigeweapon, 2021-02-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I've read through the https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup and being confused about one thing. In the External Trigger part, it said:

Pulse width = 60ms(setup) + #of frames33ms
For example: for capturing 3 consecutive frames Ton needs to be high for 160ms

Does this mean there is a 60ms setup time in EVERY lidar-on period, and only after that capturing can happen? If so, it'll be a big problem for shooting video, because it hurts the FPS too much.

RealSenseSupport, 2021-02-25:
Hi @seigeweapon 

You are understanding the information correctly and FPS will be adjusted down accordingly.

seigeweapon, 2021-02-26:
hi, Thanks for the quick response.
I don't know much about the photoelectricity that makes L515 works. But does it really need the whole 33ms for laser scanning? Some other ToF system, like Azure Kinect, uses several 100+us laser scans with 1300+us interval between, thus allowing user to set an offset of 160us between 2 synced cameras to avoid interference, while still making the 2 frames' exposure almost at the same time.
Is it possible for L515 to support similar feature in the future? Shooting multi-cam outside-in video is in hot demand. Please consider it. Thanks.

RealSenseSupport, 2021-03-23:
Yes our technology needs the 33ms for laser scanning.  

Thanks for the information and the request for future product.

closed 2021-03-23 by RealSenseSupport