realsenseai/librealsense #8449: D415 and D430 camera module mounting

GrubbyHalo, 2021-02-25:
There are two holes at the back of the D415 and D430 camera modules (One circular and one oval). Are these holes suitable for mounting ?

[image]

MartyG-RealSense, 2021-02-25:
Hi @GrubbyHalo  I researched your question carefully.  Rather than being a pair of mounting holes like the ones on the rear of the cased equivalents (D415 and D435), there are matching holes on the front of the D415 and D430 modules (see the images below).  So whilst the purpose of the holes is not defined in the data sheet document for the 400 Series cameras, it is likely that they are part of the module's technical infrastructure rather than being for the purpose of mounting the module.

[image]

[image]

On the cased D415 and D435, rear mounting holes would be used for mounting the camera by its rear instead of the tripod thread on the base of the camera.

[image]

The take-apart image of the D435 from the official RealSense website shows how the rear mounting holes go into sealed spaces rather than penetrating straight through to the module board.

[image]

GrubbyHalo, 2021-02-25:
Thanks @MartyG-RealSense for the answer. Maybe another question you might know the answer to while we are on the subject. The D430 module clearly has mounting tags on the module casing making it relatively simple to embed into a custom casing.
[image]
 
the D415 has no such tags. What is the recommended way of mounting it in a custom housing ? The D400 datasheet recommends mounting brackets as depicted here:
[image]

However using this type of bracket makes it almost impossible to mount the recommended heat-sink onto the back of the camera module.

MartyG-RealSense, 2021-02-25:
THere was a discussion about this subject in 2019.  The RealSense user in that case had the same concerns about mounting.  

https://community.intel.com/t5/Items-with-no-label/Does-the-D420-come-with-screw-mounts-Why-are-they-not-in-the-CAD/td-p/624575

At the end of the discussion, a RealSense team member from that forum suggested creating a custom 3D-printed fixing for the boards that do not have side-mounts.

GrubbyHalo, 2021-02-25:
The D415 has an IR project so will need a bracket. It seems like the bracket will work with thermal filler. Any idea where these brackets can be bought ?

MartyG-RealSense, 2021-02-25:
I recall that in earlier editions of the data sheet document (which I have just checked to confirm), a company called Faspro Technologies Inc was cited as an ordering source for brackets.

https://fasprotech.com/

agrunnet, 2021-02-25:
@GrubbyHalo Please work directly with an Intel engineer when considering mounting of modules. There are some very strict guidelines. Please do not use those holes or the pads you indicted. The central mounting bracket is the preferred way to go. 

The key thing is that you will harm the optical performance if you don’t mount properly. Some conventional considerations like thermals are secondary to the greater concerns of optical performance. Basically consider this a very sensitive scientific sensor and the way it is handled and mounting is critical. And in particular mounting stiffly and constraining module in any way is very bad. I hope that helps.

GrubbyHalo, 2021-02-25:
> I recall that in earlier editions of the data sheet document (which I have just checked to confirm), a company called **Faspro Technologies Inc** was cited as an ordering source for brackets.
> 
> https://fasprotech.com/

Thank you @MartyG-RealSense

GrubbyHalo, 2021-02-25:
> @GrubbyHalo Please work directly with an Intel engineer when considering mounting of modules. There are some very strict guidelines. Please do not use those holes or the pads you indicted. The [...]

Thank you for the advice @agrunnet .

closed 2021-02-26 by GrubbyHalo