realsenseai/librealsense #8453: Getting point clouds saved in .pts (preferably) or .las formats using Intel RealSense Lidar Camera L515

FionaBroccoly, 2021-02-26:
Hi,

I'm looking for desicions on how to get point clouds in .pts (preferably) or .las fomats using L515.

I've been reserching the subject, but have found the proper desigion neither in Intel RealSence Viewer, nor in RTABMap, nor in MeshLab. That could be because of my low skills in these softwares, or because there is a need to look for another way to fix my problem. 

So, I'm looking forward for advice and recomendation, which would be very helpful.

 P.s. Also I've reviewed the issue L515: General information/direction on handheld 3D Scanning (https://github.com/IntelRealSense/librealsense/issues/7565), where the invese problem of getting 3D Mesh (.ply, .obj) is dealt with, which has been usefull.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-02-26:
Hi @FionaBroccoly 

There are some 3D Scanning blogs on our site located here:  https://www.intelrealsense.com/category/3d-scanning/ or usage models here:  https://www.intelrealsense.com/use-cases/

One potential for you to investigate is the Dot Product:  https://www.intelrealsense.com/practical-3d-scanning-with-dot3d-pro/

FionaBroccoly, 2021-03-02:
Thank you for your answer, but of course, I've seen the sites you recoomend. The information is just very jeneral... So, I'm still looking for a specific responce to my specific question "How to get point clouds in .pts or .las fomats using L515?"

RealSenseSupport, 2021-03-23:
I understand but I'm sorry our librealsense SDK and camera does not support saving point clouds in these formats.  A 3rd party solution would have to be evaluated.

closed 2021-03-30 by RealSenseSupport