realsenseai/librealsense #8454: L515 LiDAR Camera on Raspberry Pi 4

umbnich, 2021-02-26:
Hi! 
I'm interested to use an L515 on a Raspberry Pi 4.

I've tried using the official intel realsense image for Raspberry but the device seems to be undetected. 
I've also tried with the last version of raspbian but encountered different problems during installation.

Definitely, is possible to run the L515 on a Raspberry? What operative system I've to use? 

Thank you.

RealSenseSupport, 2021-02-26:
Hi @umbnich 

Are you able to test out the L515 camera on a Windows machine?  Try using our RealSense Viewer on a Windows machine to validate that the camera functions properly.

umbnich, 2021-02-27:
Yes. I've tested on Windows and it works perfectly

autimator, 2021-03-06:
Hello @umbnich I used the raspberry pi imager to install **Raspbian** on 18 raspberry pi 4B (with 4GB of Ram)
https://www.raspberrypi.org/blog/raspberry-pi-imager-imaging-utility/

Then I followed this tutorial to install the realsense-viewer:
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

I can connect to the Lidar and take images, the pyrealsense code also works the same as on windows.
(I have to add that I only did short tests so far, and am currently struggling to synchronize the lidars  [issue: 8511](https://github.com/IntelRealSense/librealsense/issues/8511))

RealSenseSupport, 2021-03-23:
Hi @umbnich 

Were you able to get your system running with L515?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-04-02 by RealSenseSupport