realsenseai/librealsense #8455: Invalid SKU on ubuntu after firmware upgrade in windows

emaffione, 2021-02-26:
| Required Info                         |                                                                                  |
|---------------------------------|--------------------------------------------------------|
| Camera Model                       | D455                                                                         | 
| Firmware Version                   | not available after recommended upg in windows  | 
| Operating System & Version | Linux Ubuntu 18.04 LTS / Windows10                      | 
| Kernel Version (Linux Only)    |   4.15.0-136-generic                                                 | 
| Platform                                 | PC                                                                             |
| SDK Version                           |  2.0                                                                            |  
| Language                               |  C/C++/Python/Opencv                                           | 
| Segment			         |  Deep Learning                                                         | 

### Issue Description
I've upgraded a D455 on my windows system following the recommendation provided by the realviewer. Unfortunately the realsense version was not updated and something get wrong. After upgrading, the camera is not recognized both on windows and linux.
I've investigated on this behavoiour and after having verified that everything it is fine with an other D455, *not upgraded*, (so no cable/driver/software/... problem) I found that this seems due to the upgrade with a wrong version (noticeably it was the one recommended!).

On windows, plugging-in the upgraded camera the message is: "The last device you connected to this computer system malfunctioned and Windows does not recognize it". Quite hermetic...

In Linux the message is more useful in my opinion (/var/log/syslog):
Feb 25 09:10:58 DEEP-ONE kernel: [  284.426565] usb 2-1: new SuperSpeed USB device number 39 using xhci_hcd
Feb 25 09:10:58 DEEP-ONE kernel: [  284.446915] usb 2-1: config 1 interface 0 altsetting 0 has 0 endpoint descriptors, different from the interface descriptor's value: 2
Feb 25 09:10:58 DEEP-ONE kernel: [  284.447111] usb 2-1: New USB device found, idVendor=8086, idProduct=**0af5**
Feb 25 09:10:58 DEEP-ONE kernel: [  284.447116] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 25 09:10:58 DEEP-ONE kernel: [  284.447120] usb 2-1: Product: Intel(R) RealSense(TM) Invalid SKU
Feb 25 09:10:58 DEEP-ONE kernel: [  284.447124] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Invalid SKU
Feb 25 09:10:58 DEEP-ONE kernel: [  284.447128] usb 2-1: SerialNumber: 025523051986
Feb 25 09:11:03 DEEP-ONE kernel: [  289.666754] usb 2-1: Enable of device-initiated U2 failed.
Feb 25 09:11:03 DEEP-ONE mtp-probe: checking bus 2, device 39: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1"
Feb 25 09:11:03 DEEP-ONE mtp-probe: bus: 2, device: 39 was not an MTP device

Comparing the log obtained from a **functioning D455**:
Feb 25 09:22:08 DEEP-ONE kernel: [  954.150124] usb 2-1: new SuperSpeed USB device number 83 using xhci_hcd
Feb 25 09:22:08 DEEP-ONE kernel: [  954.170711] usb 2-1: New USB device found, idVendor=8086, idProduct=**0b5c**
Feb 25 09:22:08 DEEP-ONE kernel: [  954.170717] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 25 09:22:08 DEEP-ONE kernel: [  954.170721] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 455
Feb 25 09:22:08 DEEP-ONE kernel: [  954.170725] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455
Feb 25 09:22:08 DEEP-ONE kernel: [  954.170728] usb 2-1: SerialNumber: 025523051234
Feb 25 09:22:08 DEEP-ONE kernel: [  954.171828] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
Feb 25 09:22:08 DEEP-ONE kernel: [  954.173341] uvcvideo: Unable to create debugfs 2-83 directory.
Feb 25 09:22:08 DEEP-ONE kernel: [  954.173443] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Feb 25 09:22:08 DEEP-ONE kernel: [  954.173448] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
Feb 25 09:22:08 DEEP-ONE kernel: [  954.173452] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
Feb 25 09:22:08 DEEP-ONE kernel: [  954.173605] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input13
Feb 25 09:22:08 DEEP-ONE kernel: [  954.174287] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
...

the problem seems to be the fact that the upgraded camera pretend to have a IdProduct equal to **0af5** and I assume that this is due to the wrong upgrade.

Unfortunately even following the instructions provided for similar [issues ](https://github.com/IntelRealSense/librealsense/issues/5052#issuecomment-723546113) don't work and all the tools cannot see the upgraded camera because it is not binded to a correct driver => no device connected.

So the question is pretty simple: how can I override the Product ID in order to **trick** (bypass?) the system to accept the wrong PID as a D435 PID and allowing a correct firmware upgrade with the realsense tools? 
Or should I consider the camera absolutely dead? 

I think that a tool for recovering from a similar wrong state **could be really useful**

RealSenseCustomerSupport, 2021-02-26:
----------------------------------------------

The PID=0x0af5 means the RGB ISP is under FW updating. This PID is supposed to stay shortly then change to 0x0b5c. Is the camera now always showing PID=0af5?
If the D455 is unable to be recognized on USB port, that could be "dead" in FW. You probably did FW update with some old one which is not supported on D455. Do you remember which old FW version was used and caused the issue? Let me try  reproducing the problem and see if there is any solution to recover it.

Thanks

emaffione, 2021-02-27:
Thank you for your answer. 
It is still showing PID 0af5. 
Unfortunately I don't remember the version used by the recommended upgrade but I can say that the realsense version on windows was 2.34 and the original version on d455 was the one provided with the camera. Hopefully this allow you to reproduce the problem.

icarpis, 2021-02-28:
@emaffione, could you please try the following on **Linux** OS?
1. Install Python3.7-64
2. Extract [ForceDFU.zip](https://github.com/IntelRealSense/librealsense/files/6056451/ForceDFU.zip)
3. Type: _python3.7 ForceDFU.py_
4. Type: _lsusb_
5. Check whether the PID has changed.

emaffione, 2021-02-28:
Thank @icarpis 
I tryed (several times) but it doesn't work: 

$ python3.7 ForceDFU.py

No RealSense Camera has found!

I think that the problen is due to the fact that linux continously reassign a new id to the not recognized device. 
Launching lsusb more times I get different Device Id.

$ lsusb
Bus 002 Device **012**: ID 8086:0af5 Intel Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ lsusb
Bus 002 Device **013**: ID 8086:0af5 Intel Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I don't know if I can stop this behaviour...

icarpis, 2021-02-28:
@emaffione, thank you for trying my suggestion.
Yes you right, it looks that Linux continuously reassign new device id.
Maybe we have a short time window for sending the DFU mode command.
Could you please try again the above with the following new Python script?
[ForceDFU2.zip](https://github.com/IntelRealSense/librealsense/files/6057757/ForceDFU2.zip)
Also, please try to run with _sudo_, since the 0xaf5 PID is not added to the Udev Rules file.

emaffione, 2021-02-28:
Thanks for you new script.
I've launched the command as sudo and then lsusb. The PID is always 0xaf5. I think I should trick  the usb driver in order to stop reassigning a new id and then use the script.  

$ sudo python3.7 ForceDFU2.py
Try to enter DFU mode for 7 seconds...
Done!
$ lsusb
Bus 002 Device 017: ID 8086:**0af5** Intel Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

RealSenseCustomerSupport, 2021-03-01:
----------------------------------------------

I did a quick test on Viewer 2.34.0 by using Recommended FW which is 5.12.3, but the D455 is working. However one issue is found that there is no option to stream Left IR. But anyway, the Viewer is too old to D455. The latest Viewer doesn't have such issue.

Thanks

Attachment(s):
D455_RS2.34.0.PNG - https://realsensesupport.intel.com/attachments/token/eAqpPrGsIQlFAuXj5pkmtJpFl/?name=D455_RS2.34.0.PNG

emaffione, 2021-03-01:
@RealSenseCustomerSupport  Thank you for your answer. I see you cannot completely reproduce my issue, but is there any option I can follow in order to correct the problem on the upgraded D455?

RealSenseCustomerSupport, 2021-03-02:
----------------------------------------------

Hi "emaffione",

I think it's better to return this D455 and we can recover it from "dead" firmware. Please leave your email address here (format: i.e. <xxxATxxxxxDOTcom>), I will contact you via Intel email address and tell you how to return it.

Thanks

fgldmn, 2021-03-02:
I have the same issue with my device. Last week I got the notification on Windows 10 that a Firmware Upgrade was available for the D455, which I installed.

The device is now bricked. Upon connecting it to a Windows 10 PC, the message "USB device has not been recognized" pops up.

Entering the following command:
`rs-fw-update -l`
results in:
`No RealSense Camera has found!`

Does this mean it's not possible to revert this firmware "upgrade"?

RealSenseCustomerSupport, 2021-03-02:
----------------------------------------------

Hi fgldmn,

Can you please tell us which RealSense Viewer version you were using for the FW update?
If it's unable to be recognized on USB port, there is no SW tool can recover it. Please return it or leave your email address here (format: i.e. <xxxATxxxxxDOTcom>). I will contact you via Intel email address and tell you how to return it.

Thanks

fgldmn, 2021-03-02:
Thanks for getting back to me so quickly.

I installed the SDK on 31st of August 2020, the filename of the installer was Intel.RealSense.SDK-WIN10-2.38.1.2223.exe.
I cannot tell you further details because I installed the newest SDK (2.42.0) yesterday in an attempt to resolve the issue.

my email is florian[dot]goldmann[at]iis[dot]fraunhofer[dot]de

emaffione, 2021-03-02:
> Hi "emaffione",
> 
> I think it's better to return this D455 and we can recover it from "dead" firmware. Please leave your email address here (format: i.e. ), I will contact you via Intel email address and tell you how to return it.
> 
> Thanks

Thanks for your support. I'll be glad to help you in finding out a software solution, if possibile.

My email is eugenio[at]geoinference[dot]com

closed 2021-04-28 by RealSenseCustomerSupport

Sujeetdas07, 2024-07-06:
I'm also facing the same issue where the realsense D457 is not detected when connected to system (windows and linux). The issue is because the firmware has not installed properly. And currently I'm getting "Invalid SKU" when listing all usb devices.
Please assist in fixing the issue.

MartyG-RealSense, 2024-07-06:
Hi @Sujeetdas07  On the D457 model there is a hidden mode switch in the casing that needs to be moved sidewards into the 'USB' position in order to use the camera with a USB cable.  Have you moved the switch across to 'USB' in the way described at the link below, please?

https://support.intelrealsense.com/hc/en-us/community/posts/14840675121043/comments/14843232242835

Sujeetdas07, 2024-07-06:
Yes @MartyG-RealSense, the switch was always on the left side as earlier it was working fine, currently the camera is not detected in linux and getting 'Invalid SKU' as the firmware was not installed properly in windows. I tried updating the firmware using rs-fw-update, but it shows no device connected.

MartyG-RealSense, 2024-07-06:
If you have an Invalid SKU error resulting from a failed firmware update then it cannot be fixed with software, unfortunately.  The only solution is to mail your camera to Intel to have it recovered.

If you would like to proceed with that option then please send me an email at the address below with the name of your country included in it and I can provide further details.

martyx.grover@intel.com

Sujeetdas07, 2024-07-08:
Sure, I will send an email right away, also just want to clarify the statement mentioned in one thread, where firmware can be updated by opening up the realsense camera and disconnecting the vision processor D4 board. So, can you confirm if this method will work for sure (if tried) as we have a tight deadline ahead. Please let me know, but at the same let's proceed ahead in returning the camera and get to know the overall time required it to fix it from your side.

MartyG-RealSense, 2024-07-08:
@Sujeetdas07  Thanks very much.  I have replied to your email and provided in it an estimate for the time required for a fix after receiving a returned camera.

You can restore the camera yourself by replacing the Vision Processor D4 board.  The **V5** version of the D4 board must be used with a D457.

Sujeetdas07, 2024-07-08:
I just have to follow the steps mentioned in the below link right?
https://support.intelrealsense.com/hc/en-us/community/posts/16647570577683-D457-Camera-no-longer-detected-after-a-failed-change-of-firmware

MartyG-RealSense, 2024-07-08:
You can update the firmware either with rs-fw-update as described in the above link, or do it in the RealSense Viewer tool.

Sujeetdas07, 2024-07-08:
Sure, thanks for the info. will decide and revert back to you. Thanks for the assistance