realsenseai/librealsense #8456: Builing RealSense 2.42.0 with OpenVINO support results in - openvino-helpers.h:189:5: error: expected class-name before ‘{’ token

jomoengineer, 2021-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.12.11.0 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.8.0-43-generic                                         | 
| Platform                                 | Ubuntu 20.04 VirtualBox VM  |
| SDK Version                            |  2.42.0                          | 
| Language                            |  C                          | 
| Segment			|  Computer Vision                        | 

### Issue Description
While compiling RealSense 2.42.0 with OpenVINO support from source on either a Raspberry Pi 4 or an Ubuntu 20.04 VM, the build fails with the following error:

```
[ 63%] Built target rs-distance
[ 64%] Built target rs-post-processing
[ 66%] Built target rs-record-playback
Scanning dependencies of target rs-motion
[ 67%] Building CXX object examples/motion/CMakeFiles/rs-motion.dir/rs-motion.cpp.o
In file included from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/base-detection.h:7,
                 from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/object-detection.h:6,
                 from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/dnn/rs-dnn-vino.cpp:10:
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:189:5: error: expected class-name before ‘{’ token
  189 |     {
      |     ^
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:190:14: error: ‘void openvino_helpers::error_listener::onError(const char*)’ marked ‘override’, but does not override
  190 |         void onError( char const * msg ) noexcept override
      |              ^~~~~~~
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/dnn/rs-dnn-vino.cpp: In function ‘int main(int, char**)’:
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/dnn/rs-dnn-vino.cpp:219:12: error: ‘class InferenceEngine::Core’ has no member named ‘SetLogCallback’
  219 |     engine.SetLogCallback( error_listener );
      |            ^~~~~~~~~~~~~~

.
.
.

make[2]: *** [wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/build.make:63: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/rs-face-vino.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1229: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 65%] Building CXX object wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/__/rs-vino/detected-object.cpp.o
make[2]: *** [wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/build.make:76: wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/__/rs-vino/base-detection.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/build.make:63: wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/rs-dnn-vino.cpp.o] Error 1
make[2]: *** [wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/build.make:89: wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/__/rs-vino/object-detection.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1256: wrappers/openvino/dnn/CMakeFiles/rs-dnn-vino.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
 removing possibly corrupted object files 
[  1%] Built target fw
[  7%] Built target realsense-file
[ 42%] Built target realsense2
[ 47%] Built target pybackend2
[ 51%] Built target pyrealsense2
[ 51%] Building CXX object wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/rs-face-vino.cpp.o
In file included from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/base-detection.h:7,
                 from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/object-detection.h:6,
                 from /home/jovision/develop/RealSense/librealsense/wrappers/openvino/face/rs-face-vino.cpp:8:
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:189:5: error: expected class-name before ‘{’ token
  189 |     {
      |     ^
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:190:14: error: ‘void openvino_helpers::error_listener::onError(const char*)’ marked ‘override’, but does not override
  190 |         void onError( char const * msg ) noexcept override
      |              ^~~~~~~
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/face/rs-face-vino.cpp: In function ‘int main(int, char**)’:
/home/jovision/develop/RealSense/librealsense/wrappers/openvino/face/rs-face-vino.cpp:41:12: error: ‘class InferenceEngine::Core’ has no member named ‘SetLogCallback’
   41 |     engine.SetLogCallback( error_listener );
      |            ^~~~~~~~~~~~~~
make[2]: *** [wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/build.make:63: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/rs-face-vino.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1229: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```
The CMake command used is:
```
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=$(which python3) -DFORCE_LIBUVC=true -DBUILD_OPENVINO_EXAMPLES=true -DOpenCV_DIR=/opt/intel/openvino_2021/opencv/cmake -DINTEL_OPENVINO_DIR=/opt/intel/openvino_2021 .. 
```

MartyG-RealSense, 2021-02-27:
Hi @jomoengineer   As a starting point in investigating your case: if you need to use a VM with librealsense then Intel recommend using **VMWare Workstation Player** instead of Oracle VirtualBox in order to provide proper emulation of the USB 3 controller.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

jomoengineer, 2021-02-27:
@MartyG-RealSense So that explains why the build is failing for the Raspberry PI as well?

MartyG-RealSense, 2021-02-27:
It might be an influencing factor but I would not like to say for sure without knowing more details about your Pi 4 setup.  For example, when using Ubuntu with Pi instead of Raspbian, the last couple of cases I have handled that have been successful have involved Ubuntu **Server**.  This subject is discussed in the recent ROS-related case below:

https://github.com/IntelRealSense/realsense-ros/issues/1724#issuecomment-786508280

jomoengineer, 2021-02-27:
@MartyG-RealSense 
 The links you reference are not related to the issue I am seeing.

The Raspberry Pi I am using is running
  Raspbian OS 10 (buster) 
  Kernel: 5.10.11-v7l+

The VirtualBox VM that is hosting Ubuntu 20.04 has USB3 support enabled with the RealSense device filters added.

Note, I can build, install and run both OpenVINO ToolKit 2021.2.185 and RealSense 2.42.0 (Without OpenVINO support) individually on both the Raspbian and Ubuntu VM config and the RealSense camera is indented and usable in both the Raspbian and Ubuntu instances.  The issue I am having, which is what I prefer to focus on in this issue, is that I see the posted error when attempting to build RealSense Release v2.42.0 (or v2.41.0) with OpenVINO support as listed in the RealSense OpenVINO Wrapper page at the bottom under Linux:
[https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino](url)

I am curious if anyone else is successful in building RealSense with OpenVINO support as shown in the Linux example?

MartyG-RealSense, 2021-02-27:
I note that you are using **OpenVINO ToolKit 2021.2.185**.  Compatibility up to OpenVINO 2020 was added in SDK 2.41.0.  

https://github.com/IntelRealSense/librealsense/pull/7851

I would therefore recommend using OpenVINO **2020** with the OpenVINO wrapper examples if it is possible for you to downgrade.

jomoengineer, 2021-02-27:
@MartyG-RealSense 
Well, this seems to run on Windows 10, so I am not sure why there is a limitation with Linux. I have no interest in downgrading to a 2020 version of OpenVINO and will just write this up in my review as not being supported with Linux and works in Windows only.
Also, I will have a word or two with regards to the Intel support and this product.

You can close this issue. I have no interest in pursuing it any further and will seek another vendor option.

MartyG-RealSense, 2021-02-27:
I'm sorry that you have not had a smooth experience with your OpenVINO implementation.  I will feed your comments back to Intel.

jomoengineer, 2021-02-28:
Just a note, with OpenVINO 2020.3 on the Raspberry Pi, it gets different error, although I believe this is due to ngraph not being ported on the Raspberry Pi.
```/usr/bin/ld: /opt/intel/openvino/opencv/lib/libopencv_dnn.so.4.3.0: undefined reference to `ngraph::Node::get_input_size() const'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `typeinfo for ngraph::op::v0::TensorIterator'
/usr/bin/ld: /opt/intel/openvino/opencv/lib/libopencv_dnn.so.4.3.0: undefined reference to `ngraph::as_node_vector(std::vector<ngraph::Output<ngraph::Node>, std::allocator<ngraph::Output<ngraph::Node> > > const&)'
/usr/bin/ld: /opt/intel/openvino/opencv/lib/libopencv_dnn.so.4.3.0: undefined reference to `ngraph::Node::get_input_element_type(unsigned int) const'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `typeinfo for ngraph::op::v0::RegionYolo'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `ngraph::op::v0::Tile::Tile(ngraph::Output<ngraph::Node> const&, ngraph::Output<ngraph::Node> const&)'
/usr/bin/ld: /opt/intel/openvino/opencv/lib/libopencv_dnn.so.4.3.0: undefined reference to `ngraph::Node::description[abi:cxx11]() const'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `typeinfo for ngraph::op::v1::ReduceLogicalOr'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `typeinfo for ngraph::op::v0::Abs'
/usr/bin/ld: //opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine_transformations.so: undefined reference to `ngraph::pass::ConstantFolding::construct_constant_range()'
/usr/bin/ld: //opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine_transformations.so: undefined reference to `ngraph::op::v1::Pad::get_pads_begin() const'
/usr/bin/ld: /opt/intel/openvino/opencv/lib/libopencv_dnn.so.4.3.0: undefined reference to `ngraph::Node::m_next_instance_id'
/usr/bin/ld: /opt/intel/openvino/inference_engine/lib/armv7l/libinference_engine.so: undefined reference to `typeinfo for ngraph::op::v0::Sigmoid'
collect2: error: ld returned 1 exit status
make[2]: *** [wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/build.make:164: wrappers/openvino/face/rs-face-vino] Error 1
make[1]: *** [CMakeFiles/Makefile2:372: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

MartyG-RealSense, 2021-02-28:
Thanks very much for your patience and the additional information, which will be included in discussion with Intel about your problems in this case.

Given that the RealSense OpenVINO wrapper makes use of pre-trained models from the Open Model Zoo and the link below states that the Zoo needs to be downloaded separately on Raspberry Pi, it may be worth trying that Zoo download.

https://www.intel.co.uk/content/www/uk/en/support/articles/000055510/boards-and-kits/neural-compute-sticks.html

jomoengineer, 2021-02-28:
@MartyG-RealSense 
Yeah, I am quite aware of Open Model Zoo and have been running through the demos on the Raspberry Pi and Ubuntu 20.04 x86_64.  However, again, I have been using the 2021.2 Release to keep inline with the 2021.2 Release of OpenVINO I have been using.  The link you reference is for 2020.3 release.
The non UK one is:
[https://www.intel.com/content/www/us/en/support/articles/000055510/boards-and-kits/neural-compute-sticks.html](url)

These are the links I have been following for OpenVINO and OMZ:
[https://docs.openvinotoolkit.org/latest/openvino_docs_install_guides_installing_openvino_raspbian.html](url)
[https://docs.openvinotoolkit.org/latest/omz_demos_README.html](url)

I have previously posted an issue with regards to ngraph and bulding the OMZ Python Demos on the Raspberry Pi 4:
[https://github.com/openvinotoolkit/open_model_zoo/issues/2135](url)

I'll have to look through the config and see if I can remove references to YOLO and then see if it builds.
Also, I did not see any reference for placement of OMZ for RealSense on a Linux build, so I'll to look for that as well if it is in fact a dependency.

MartyG-RealSense, 2021-02-28:
Thanks very much @jomoengineer for the detailed update - I look forward to your next report after the tests described above.  Good luck!

MartyG-RealSense, 2021-03-08:
Hi @jomoengineer  Do you have an update about your case that you can provide, please?   Thanks!

jomoengineer, 2021-03-08:
@MartyG-RealSense Are you with Intel?  I was hoping that someone from Intel would chime in on this.

I complete my review of the product but would like to find a solution for this issue.  My preference would be to use the 2021.2 version rather than the older 2020 Release since there are differences in the demo naming structure and use.

MartyG-RealSense, 2021-03-08:
I am a RealSense support engineer with Intel, yes.

At the present time, support is offered up to OpenVINO 2020.  I do not have information regarding support for newer OpenVINO versions at a future date.

jomoengineer, 2021-03-08:
Okay.  You can close this. I have not interest in pursuing this further.  I'll just mark it us unsupported.

closed 2021-03-08 by MartyG-RealSense