realsenseai/librealsense #8457: depth image transformation

fisakhan, 2021-02-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.11.01.100) | 
| Operating System & Version |   {Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  (5.4.0-65-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<40>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
The realsense D435 is fixed on a wall and I want to rotate the captured depth image around Y axis (center of image). Please see here an example of image and rotated image in pixel domain. Given the angle of rotation, I want to get the rotated depth image with relevant depths/distances from camera. The depths in the rotated image should look like as someone rotated the object/scene.
[image]
[image]

MartyG-RealSense, 2021-02-27:
Hi @fisakhan  My understanding is that whilst it is straightforward to rotate the image, it can have consequences for the intrinsics and extrinsics, which will also need to be updated if you plan to use the information in the image.

https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-597615585

https://github.com/IntelRealSense/librealsense/issues/4395#issuecomment-510744764

A RealSense user recently shared on a case about image rotation their method for how they perform the processing of the image in the **Open3D** library.

https://github.com/IntelRealSense/librealsense/issues/6023#issuecomment-785071263

MartyG-RealSense, 2021-03-06:
Hi @fisakhan  Do you require further assistance with this case, please?  Thanks?

closed 2021-03-12 by MartyG-RealSense