realsenseai/librealsense #8458: Sync L515 depth frame with moving conveyor using external trigger

tomloomans88, 2021-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 |
| Firmware Version                   | 1.5.3.0 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC  |
| SDK Version                            |  2.40.032482                          | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Setup Description
I have a multi-camera l515 setup (4 cameras) above a moving conveyor belt. I've connected the cameras electrically (using the [Multi-Camera configurations whitepaper](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) as an example) to an [Arduino pro micro 3.3V](https://www.sparkfun.com/products/12587) pin A0 to A3 to trigger them one after another (they are aimed in the same direction and I was getting interference when they were triggering at the same time).

I'm using this very basic Arduino code to trigger the cameras.

    void setup() {
      pinMode(A3, OUTPUT);
      pinMode(A2, OUTPUT);
      pinMode(A1, OUTPUT);
      pinMode(A0, OUTPUT);  
    }
    
    void loop() {
      
      // Trigger cam 1
      digitalWrite(A3, LOW);
      delay(150);   
      // Stop cam 1  
      digitalWrite(A3, HIGH);
      delay(30);   
    
      // Trigger cam 2  
      digitalWrite(A2, LOW);
      delay(150);   
      // Stop cam 2 
      digitalWrite(A2, HIGH);
      delay(30);  
    
      // Trigger cam 3 
      digitalWrite(A1, LOW);
      delay(150);   
      // Stop cam 3
      digitalWrite(A1, HIGH);
      delay(30);  
    
      // Delay the loop
      delay(200);
    }

### Issue Description

Using the 150ms trigger time, I get either 0, 1 or 2 frames each time I trigger a camera. Since my objects are moving on a conveyor and the cameras are triggered after each other, the object has moved between frames. Therefore I need to know exactly when the image is taken so I can latch my rotary encoder counter connected to the conveyor at each image capture so I can compensate later for the movement of the object.

The problem is that I do not know when exactly the image is taken. The whitepaper mentioned a setup time of 60ms after which the camera starts taking images. The fact that I'm getting inconsistent amounts of frames gives me the idea that this setup time is not deterministic. This means that somewhere in my 150ms trigger time an image is captured, but I have no way to exactly know when. Using the timestamp of the frame doesn't help because my encoder counter board is not on the same timer as the camera.

Is there a way to get an electrical signal out of the camera at the exact time it makes a frame? Or is there a way to get the camera to trigger always after a same delay after the trigger signal is set?

Also I couldn't fine anything about this trigger port in the manual (is it opto isolated? what is the maximum voltage it can handle? etc...)

I really hope someone can give me some advice on this issue.

RealSenseSupport, 2021-03-22:
We are looking at changes to the whitepaper to help explain the solution better.  We will notify when the update has been done.

RealSenseSupport, 2021-04-26:
Hi @tomloomans88 

Our team has made significant updates to the whitepaper.  Please check the updated whitepaper here:  https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

tomloomans88, 2021-04-28:
Thank you for updating the whitepaper, it is cloud now how the trigger mechanism works. However, it does mean that I won't be able to sync the camera with a moving conveyor since the time between the trigger enable and the first full frame is not constant. I will have to look for a different camera to do that for this setup.

Do you think future models of the realsense product line will have a trigger mechanism which ensures an immediate frame capture after the trigger is set?

RealSenseSupport, 2021-05-10:
We are continuing to look at ways to improve our future products.  We will take this feedback and look at future products.