realsenseai/librealsense #8465: "No frame receiver" status on intel realsense viewer in PC

nh-97, 2021-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer  v2.41.0) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| Language                            |  {C/C# }                          | 

### Issue Description
I have a problem using the L515 camera. When the intel Realsense viewer is turned on the depth, IR, confidence streams in the depth sensor fail (No frame receiver!) for 1-2 seconds, then it works normally. But with RGB sensor, it works well
[image]

nh-97, 2021-03-10:
@MartyG-RealSense 
Sorry, Can you help me with this problem?

RealSenseSupport, 2021-03-10:
Hi @nh-97 

Have a couple of questions based on what you're seeing:
- Are you seeing the message only at initial enabling of the streams?  I'm thinking its the initial frames until the device gets up and running.
- I notice that the RS Viewer you're using is 2.38.1, are you seeing this same issue on the latest Viewer 2.42?
- What FW do you have on your device?  Can you show the "info" in the Viewer?
- Also provide the settings in which you have for the depth streams?

nh-97, 2021-03-11:
Thanks @RealSenseSupport !

> Chào @ nh-97
> 
> Có một số câu hỏi dựa trên những gì bạn đang thấy:
> 
> * Bạn có thấy thông báo chỉ khi bật luồng ban đầu không? Tôi đang nghĩ về những khung hình ban đầu cho đến khi thiết bị khởi động và chạy.
When streaming it does not show any message. Yep, don't get the first frames.
> * Tôi nhận thấy rằng Trình xem RS bạn đang sử dụng là 2.38.1, bạn có gặp sự cố tương tự trên Trình xem 2.42 mới nhất không?
I have updated RS Viewer to v2.40.1, v2.41.0, v2.42.0 but still have this problem. Also update the Firmware accordingly..
> * Bạn có FW nào trên thiết bị của mình? Bạn có thể hiển thị "thông tin" trong Trình xem không?
This is information about the device
[image]

> * Đồng thời cung cấp các cài đặt mà bạn có cho các luồng chiều sâu?
Are you wanting to know about these settings?
[image]

Besides, I have checked the usb caple (replace another usb cable 3.1 and another usb port), connect the camera to another PC. But that doesn't change the problem.

nh-97, 2021-03-11:
> Hi @nh-97
> 
> Have a couple of questions based on what you're seeing:
> 
> * Are you seeing the message only at initial enabling of the streams?  I'm thinking its the initial frames until the device gets up and running.
When streaming it does not show any message. Yep, don't get the first frames.
> * I notice that the RS Viewer you're using is 2.38.1, are you seeing this same issue on the latest Viewer 2.42?

I have updated RS Viewer to v2.40.1, v2.41.0, v2.42.0 but still have this problem. Also update the Firmware accordingly..
> * What FW do you have on your device?  Can you show the "info" in the Viewer?
This is information about the device
[image]
> * Also provide the settings in which you have for the depth streams?
Are you wanting to know about these settings?
[image]

Besides, I have checked the usb cable (replace another usb cable 3.1 and another usb port), connect the camera to another PC. But that doesn't change the problem.

nh-97, 2021-03-11:
> When streaming it does not show any message. Yep, don't get the first frames.
>
> I have updated RS Viewer to v2.40.1, v2.41.0, v2.42.0 but still have this problem. Also update the Firmware [...]

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense