realsenseai/librealsense #8469: T265 Android support

nsreedharbabu, 2021-03-01:
Hi, The T265 is not working in Android. Possibly when will RealsenseSDK2.0 start supporting android devices. Please let me know .Thanks

kinectsources, 2021-03-11:
Which Realsense product is best for Augmented Reality  D400 series or T265 camera in android ?

RealSenseSupport, 2021-03-16:
@nsreedharbabu Sorry, we don't have plan for T265 Android support. 
@kinectsources As we don't have plan for T265 Android support, please try D400 series if you tend to use it in Android. Thanks!

hectorC, 2021-03-22:
> @nsreedharbabu Sorry, we don't have plan for T265 Android support.
> @kinectsources As we don't have plan for T265 Android support, please try D400 series if you tend to use it in Android. Thanks!

I've been waiting for Android support for the T265 since long time. It's a bit confusing because you are saying there's no plan for it but the release notes all this time (including the latest) have said it's planned: "Android support is planned but not yet included"

kinectsources, 2021-03-22:
This guy inter connected to PC with his T265 and that AR is dope 

https://youtu.be/RYjvAsAw_-E

nsreedharbabu, 2021-03-22:
is it like he is using mobile back camera and T265 tracking?

RealSenseSupport, 2021-04-06:
@hectorC Thank you for pointing out the inconsistency! We're trying to make these aligned. As stated in https://www.intelrealsense.com/download/16547/?_ga=2.255863112.2142278820.1617681544-69693576.1578898374  we don't have plan for T265 Android support. Sorry for any inconvenience.

RealSenseSupport, 2021-04-09:
@hectorC Any other questions about this? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-04-21:
@hectorC No other questions from your side. Closed this ticket accordingly. Thanks!

closed 2021-04-21 by RealSenseSupport

TheBricktop, 2021-04-27:
Yet current release have this misleading information about Android support 2.43.0