realsenseai/librealsense #8471: Capturing Single Images without a Stream

mmatl, 2021-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Is it possible to trigger single-frame captures for the D415 cameras? Basically, the USB bandwidth requirement is quite high, and I only need a frame every few seconds, so I'd like to only capture frames on-demand.

MartyG-RealSense, 2021-03-01:
Hi @mmatl   If you are able to use C++,  an adaptation of the SDK example **rs-save-to-disk** may suits your needs.  When launched, it runs for about 1 second and stores 30 frames, and then writes the 30th frame to disk as png and csv files.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

A Python script in the link below takes an alternate approach of saving frames to file only when a key is pressed.

https://github.com/IntelRealSense/librealsense/issues/8388#issuecomment-782395443

MartyG-RealSense, 2021-03-08:
Hi @mmatl  Do you require further assistance with this case, please?  Thanks!

mmatl, 2021-10-06:
@MartyG-RealSense Sorry for re-opening this with such a delay. Hope you're doing well. My question is more specifically this -- is there a way to read single frames on-demand from a RealSense D4XX camera without opening a full stream? I can obviously just grab the frame I want from a stream, but the fundamental problem is bandwidth. If I have five cameras streaming at once, that eats a ton of bandwidth, and if I'm ignoring all but one frame every 3-4 seconds, the vast majority of that bandwidth is wasted. Instead, I'd like to send a software signal to the RealSense to capture a frame and send it to me instead of opening a video stream -- basically, using the camera as a still photo camera instead of a video camera. Is that possible?

eszabo12, 2022-09-20:
For anyone wondering, I created a more intuitive interface to capture images as needed:

https://github.com/eszabo12/image_capturing_pipeline