realsenseai/librealsense #8472: How to convert z16 to uint8 for face live detection？

lawo123, 2021-03-02:
I am using SR300.I want to collect depth data for live detection, but how to convert the depth map z16 to uint8?

MartyG-RealSense, 2021-03-02:
Hi @lawo123   What you are describing sounds like converting a RealSense depth image from **rs2::frame** to an OpenCV **mat** image.  If that is what you are aiming to do then the code snippet in the link below may be helpful.

https://dev.intelrealsense.com/docs/opencv-wrapper#section-1-minimal-opencv-application-for-visualizing-depth-data

If you are aiming to achieve a different goal to the one above then please provide further information.  Thanks!

lawo123, 2021-03-02:
> Hi @lawo123 What you are describing sounds like converting a RealSense depth image from **rs2::frame** to a **cv::mat** image. If that is what you are aiming to do then the code snippet in the link below may be helpful.
> 
> https://dev.intelrealsense.com/docs/opencv-wrapper#section-1-minimal-opencv-application-for-visualizing-depth-data
> 
> If you are aiming to achieve a different goal to the one above then please provide further information. Thanks!

thank you for your reply.Is there only one way to convert z16 to uint8 through the rs2::colorizer function?

lawo123, 2021-03-02:
I am looking at the data set of caisa-surf, he used the depth map collected by sr300, and the depth map I converted with the colorizer function is not the same.
[image]

MartyG-RealSense, 2021-03-02:
Looking at the details of **CAISA-SURF** in a paper, according to section 3.2 of that paper they obtained the above face depth-image by using an 3D face reconstruction tool called PRNET rather than streaming a completed depth image from the camera directly.

**CAISA-SURF paper**
https://arxiv.org/pdf/1812.00408.pdf

**PRNET**
https://github.com/YadiraF/PRNet

lawo123, 2021-03-02:
> Looking at the details of **CAISA-SURF** in a paper, according to section 3.2 of that paper they obtained the above face depth-image by using an 3D face reconstruction tool called PRNET rather than [...]
Thank you for your reply.I think prnet is only used to generate a mask, 0 or 1, and does not affect the value of the face.
[image]

MartyG-RealSense, 2021-03-02:
Comparing your face image with the CASIA-SURF one, I do not believe that there is essentially anything wrong with your depth image.  It is just more detailed.

lawo123, 2021-03-02:
> Comparing your face image with the CASIA-SURF one, I do not believe that there is essentially anything wrong with your depth image. It is just more detailed.

But why is mine more detailed? I hope to be consistent with his output.

MartyG-RealSense, 2021-03-02:
It is possible they were using the old **RealSense SDK 2016 R2** for Windows to capture the data instead of the current **RealSense SDK 2.0 (librealsense2)**, in which case it would be difficult to achieve the same style of output.  The CASIA-SURF paper states that their dataset was produced in 2018 and they used **RealSense SDK**.

A different face recognition paper linked to below, which used 2016 R2, features facial images similar to yours.

https://essay.utwente.nl/75772/1/Manterola_BA_SCS.pdf

[image]

lawo123, 2021-03-02:
> It is possible they were using the old and now-obsolete **RealSense SDK 2016 R2** for Windows to capture the data instead of the current **RealSense SDK 2.0 (librealsense2)**, in which case it would be difficult to achieve the same style of output from the current SDK (which has a completely different architecture to 2016 R2). The CASIA-SURF paper indicates that their dataset was produced in 2018 and they used **RealSense SDK**.
> 
> A different SR300 face recognition paper linked to below, which used 2016 R2, features facial images similar to yours.
> 
> https://essay.utwente.nl/75772/1/Manterola_BA_SCS.pdf
> 
> [image]
> 
> It is also possible that the CASIA-SURF dataset was produced in **Legacy Librealsense (librealsense1)**, the predecessor of the current SDK that has more in common with the current one. Legacy Librealsense is also SR300-compatible. The paper below uses Legacy Librealsense for images closer to the CASIA-SURF ones.
> 
> https://essay.utwente.nl/74345/1/Delaere_BSA_EWI.pdf
> 
> [image]
> 
> The above selection of papers and images demonstrates that there are a range of different methods and tools that can be used to achieve facial detection / recognition and there is not one single correct approach.

This image is not like the casisa-surf image. casia-surf looks like it uses the colorizer function (histogram equalization enable), so the whole picture has a clear difference between bright and dark, but why the real person’s face is so bright, there is no detailed information, my fake picture and his Is very similar.
真人：
[image]
假人：

lawo123, 2021-03-02:
> It is possible they were using the old and now-obsolete **RealSense SDK 2016 R2** for Windows to capture the data instead of the current **RealSense SDK 2.0 (librealsense2)**, in which case it would [...]

This image is not like the casisa-surf image. casia-surf looks like it uses the colorizer function (histogram equalization enable), so the whole picture has a clear difference between bright and dark, but why the real person’s face is so bright, there is no detailed information, my fake picture and his Is very similar.
cassia-surf 真人：
[image]

cassia-surf 假人：
[image]

MartyG-RealSense, 2021-03-02:
My advice is to not try to force your depth output to fit what somebody else's project looks like.  Instead, pick an approach that works for you and follow it, not worrying about what other projects have done.

lawo123, 2021-03-02:
> My advice is to not try to force your depth output to fit what somebody else's project looks like. Instead, pick an approach that works for you and follow it, not worrying about what other projects have done.

Because I want to use his data set, I don’t want to re-collect the data set.

MartyG-RealSense, 2021-03-02:
I think there may have been some confusion in this case.   The above advice has been based on the idea that you wanted to use your SR300 to create new face images with your SR300 that looked like the CASIA-SURF dataset.   Do you actually mean that you don't want to create new images and plan to just use the existing CASIA-SURF face images, please?

Going back to the start of this case, it sounds as though you want to collect live depth data of the face with the SR300 and convert it to UINT8 format, and **then** compare that UINT8 data to the existing CASIA-SURF daraset images.  Is that correct please?

lawo123, 2021-03-02:
> I think there may have been some confusion in this case. The above advice has been based on the idea that you wanted to use your SR300 to create new face images with your SR300 that looked like the CASIA-SURF dataset. Do you actually mean that you don't want to create new images and plan to just use the existing CASIA-SURF face images, please?
> 
> Going back to the start of this case, it sounds as though you want to collect live depth data of the face with the SR300 and convert it to UINT8 format, and **then** compare that UINT8 data to the existing CASIA-SURF dataset images. Is that correct please?

I want to use the casia-surf data set for model training and then use the data of sr300 to directly predict. Now the prediction effect is not good. I suspect it is caused by a problem with the depth z16 to uint8

MartyG-RealSense, 2021-03-02:
You might obtain more reliable results if you switch to using the SDK's **dlib** face landmark detection example.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib/face

With this example, you can use your own dataset or download the 68-point trained model file from dlib's site.

closed 2021-03-14 by MartyG-RealSense