realsenseai/librealsense #8474: Recreating a rs2::depth_frame from the bytes data of the frame and query frame configurations from realsense camera

sandeshk1, 2021-03-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | PC (corei7)  |
| SDK Version                            |  2.41.0                         | 
| Language                            |  C++                          | 
| Segment			|  Computer Vision                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
1. The get_data() member function of the rs2::frame class retrieves data from frame handle. In an application I want to send only the frame data in bytes format over a message bus. Now at the other end of the application I want to again create a rs2::depth_frame with this frame data. I want to recreate just the depth frame and not the entire frameset. Is there any APIs for this ?
2. How to query the maximum width, height and framerate supported by the realsense camera through the APIs ?

MartyG-RealSense, 2021-03-02:
Hi @sandeshk1  

1.  A possible approach to this program may be to convert rs2::frame to cv::mat and then at the other end convert cv::mat to rs2::frame.  Conversion is a subject that we have discussed previously.

https://support.intelrealsense.com/hc/en-us/community/posts/360052840153-Converting-Depth-Frame-to-Color-Frame-With-RealSense-Depth-Cameras

**Convert rs2::frame to cv::mat**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/imshow

**Convert cv::mat to rs2::frame**
https://github.com/IntelRealSense/librealsense/issues/2634

2.  Though you can query available stream profiles - a principle demonstrated by the **rs-sensor-control** example - I am not aware of a mechanism to detect the **maximum** setting available on a particular camera model.

https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

sandeshk1, 2021-03-02:
@MartyG-RealSense converting cv::mat to rs2::frame may be one of the approaches. 

I was trying to understand if there is a simpler way to construct a rs2::depth_frame by directly using the bytes data of the depth frame. 

For instance opencv supports Mat constructor to create a cv mat frame from the bytes data of the depth frame.

MartyG-RealSense, 2021-03-02:
Some Python users have explored the possibility of converting rs2::frame to a numpy array and back again, though if your project must use C++ then that would not be a viable option for you.

https://support.intelrealsense.com/hc/en-us/community/posts/360047819294-Conversion-of-Frame-to-Numpy-Array-and-Vice-Versa

The SDK supports using C++ to convert a depth image to a jpeg and then recover the depth image from the jpeg.  This may be an awkward process though, as the depth recovery part of the process has to be performed outside of the SDK.

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras#section-3-2-depth-image-recovery-from-colorized-depth-images-in-c

Another option might be saving the depth frame as a **raw** format image file (an image format that preserves depth information) and then retrieve the depth data from the raw by loading it into the SDK's **software_device** interface.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412810987

MartyG-RealSense, 2021-03-10:
Hi @sandeshk1  Do you require further assistance with this case, please?   Thanks!

sandeshk1, 2021-03-15:
@MartyG-RealSense This is good info for now. Will close it for now.

closed 2021-03-15 by sandeshk1