realsenseai/librealsense #8476: Exceptions are not caught.

MBSadeghzadeh, 2021-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { 2}                          | 
| Language                               |  {C++ }                          | 

### Issue Description
```
try
{
 if(camera != nullptr)
camera->pipe.stop();
}
catch (const rs2::error &e)
{
cerr << e.what();
}
```
this block of code raises an exception but I can not catch it. My project has been written in C++, Qt and CMake and is compiled by MSVC 14.

MartyG-RealSense, 2021-03-02:
Hi @MBSadeghzadeh   Does the block of catch code below make a difference to your catch success please?

```
catch (const rs2::error & e)
{
std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;		
return EXIT_FAILURE;
}
```

MBSadeghzadeh, 2021-03-03:
it does not make any difference. I get this message from visual studio:
```
Exception thrown at 0x00007FFA6F2DE66B (realsense2d.dll) in GUI.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.

If there is a handler for this exception, the program may be safely continued.
```

this exactly happens at `camera->pipe.stop();` when I don't have any camera to connect to and `camera->pipe.start(camera->cfg)` has already raised an exception which has been caught.

MartyG-RealSense, 2021-03-03:
I wonder if your use of MSVS 14 is a factor.  I would consider Visual Studio 2015 as a minimum recommendation for use with librealsense to ensure compatibility with  **C++11**.

MartyG-RealSense, 2021-03-03:
I am reminded of a small number of past cases that had an error that occurred when using pipe.stop().  Even if they successfully caught the error, the program became frozen after the stop was initiated.  A solution that worked for some of those users was to use **sensor.stop()** instead of pipe.stop (i.e start the pipeline and stop the sensor).

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

MBSadeghzadeh, 2021-03-04:
using this line `camera->pipe.get_active_profile().get_device().first<rs2::depth_sensor>().stop();` to stop the sensor,  unfortunately  it does not work and just shifted error to below function in `rs_pipeline.hpp` at line ` auto p = std::shared_ptr<rs2_pipeline_profile>(
                rs2_pipeline_get_active_profile(_pipeline.get(), &e),
                rs2_delete_pipeline_profile);`
```
pipeline_profile get_active_profile() const
        {
            rs2_error* e = nullptr;
            auto p = std::shared_ptr<rs2_pipeline_profile>(
                rs2_pipeline_get_active_profile(_pipeline.get(), &e),
                rs2_delete_pipeline_profile);

            error::handle(e);
            return pipeline_profile(p);
        }
```

Remember that it happens when I don't have any camera plugged to pc.

MartyG-RealSense, 2021-03-04:
I examined your case carefully again from the start.  I noticed a reference in your exception message to **realsense2d.dll**.  Note that it is **2d** and not 2 (as in **realsense2.dll**).  I recall a past case where this inserted 'd' in the DLL name caused problems.  

My recollection is that this filenaming may occur if librealsense has been built in **Debug** mode instead of **Release**  mode.

https://github.com/IntelRealSense/librealsense/issues/6489#issuecomment-652483312

https://github.com/IntelRealSense/librealsense/issues/5022#issuecomment-567616133

MBSadeghzadeh, 2021-03-07:
Instead of `camera->pipe.get_active_profile().get_device().first<rs2::depth_sensor>().stop();` I used `camera->device.first<rs2::depth_sensor>().stop();` and it fortunately caught the exception as `device` was empty because the `start` method had not been executed and it had raised an exception. 
Thanks @MartyG-RealSense, your comments were really helpful.

closed 2021-03-07 by MBSadeghzadeh