realsenseai/librealsense #8478: CUDA links, instructions?

arothenberg, 2021-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
Is there a step by step for setting up CUDA?
I read that I should  cmake from a CUDA file? I've been cmaking from the root I think and everything works great. How should I do it for this case? 

Also, I read that once the project is made(maked) using the  Cuda config there is nothing more that needs to be done, is that right? IOW the code I write stays the same.

MartyG-RealSense, 2021-03-02:
Hi @arothenberg   If you are installing librealsense from **packages** then CUDA support is built into them.

If you are building librealsense from source code with CMake then you can enable CUDA support by including the build flag **-DBUILD_WITH_CUDA:bool=true** in the CMake instruction.

When CUDA support is enabled, SDK operations that can benefit from CUDA (pointcloud, alignment and color conversion) should be automatically accelerated.

https://github.com/IntelRealSense/librealsense/pull/1866

arothenberg, 2021-03-02:
ok thanks

arothenberg, 2021-03-02:
I need the developer board, not just an nvidea card like a Gforce? 
Also does it affect the standard texture rendering or just the point cloud rendering.

closed 2021-03-02 by arothenberg

arothenberg, 2021-03-02:
By color conversion, you mean the color mapping  of the depth  data?

MartyG-RealSense, 2021-03-03:
As far as I know, any detectable Nvidia GPU, including video cards, is usable by librealsense's CUDA support.

The color conversion is about converting a color image from the native image format (YUY2) to another.  The formats that have support for **conversion speed acceleration** include those in the table linked to earlier.

https://github.com/IntelRealSense/librealsense/pull/1866#issue-193946710

arothenberg, 2021-03-03:
Ah ok. thanks.

arothenberg, 2021-03-03:
Wow, there is almost nothing on realsense and cuda. I'll make it and see what happens. Are there a list of cuda functions?

MartyG-RealSense, 2021-03-03:
If CUDA support is enabled and an Nvidia GPU is present then you do not need a list of CUDA functions, as the accelerations are applied automatically.

*****************

It is different for the SDK's alternate graphics acceleration method, which is called **GLSL processing blocks**.  With GLSL, you have to amend script instructions to take advantage of the acceleration - for example, changing **rs2::align** to **rs2::gl::align** or rs2::pointcloud to rs2::gl::pointcloud

The link below has a very good pros and cons analysis of GLSL processing blocks and when it can be used.

https://github.com/IntelRealSense/librealsense/pull/3654