realsenseai/librealsense #8480: Bag File information regarding IMU data

jimmek, 2021-03-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.10.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.0.2657                        | 
| Language                            |  C/C++ }                          | 
| Segment			|  other - part of data acquisition system                          | 

### Issue Description
Information for extraction of D445 IMU data from a Bag File.

Hello MartyG,

Thank you for your assistance in the past.

I have reviewed the documents "Bags_Format_1.2 - ROS Wiki" and "Bags_Format_2.0 - ROS Wiki" referenced in previous discussions and have been intently examining the bag file content for a simple Viewer data collection of only IMU data. I do not yet have a complete understanding of the bag file format, structure, and content.  I find the above two documents not descriptive enough to easily decipher all the information within the Bag file. What I am needing is Bag file information that is more descriptive so that a full understanding of the bag file can be utilized for successful extraction of the D455 IMU data. to an ASCII file. Can you or your colleagues provide such information?    Thank you.

MartyG-RealSense, 2021-03-02:
Hi @jimmek  The SDK has a C++ tool available called **rosbag-inspector** for examining the contents of a bag file in detail.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

jimmek, 2021-03-04:
Hello MartyG.

Thank you for the information regarding rosbag-inspector. I will work with my colleague in generating an executable of the rosbag-inspector program as I have not yet found an executable of it on the internet.

I was wondering if you could provide some addition information regarding Bag file format? In looking at the Bags_Format documents, there is a reference to "unique connection ID" that is not defined as to what it applies to. In looking at my sample file for IMU data there are unique ID's between 0-92. It looks as if the accelerometer data is related primarily to one connection ID value with a just a few cases with two other connection ID values. A clear definition of what a "unique connection ID" is and how it is related to various data elements/records used would be very useful.  Thank you.

MartyG-RealSense, 2021-03-04:
The bag format that the SDK uses is mostly based on the **rosbag** standard, though has some differences as the SDK bag format is only designed to handle camera data.

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

Extensive information about the SDK bag structure can be found in the official **record and playback** documentation in the link below.  It does not have information about a "unique connection ID" though.

https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md#rosbag

The SDK has a **bag.cpp** file that provides references to connection indexes.  You can highlight them by visiting the script at the link below and doing a browser 'find' operation on the page for the word **connection**.

https://github.com/IntelRealSense/librealsense/blob/master/third-party/realsense-file/rosbag/rosbag_storage/src/bag.cpp

MartyG-RealSense, 2021-03-12:
Hi @jimmek  Do you require further assistance with this case, please?  Thanks!

jimmek, 2021-03-15:
Please leave open. I have been diverted to other priority work and will get back to this issue soon.

MartyG-RealSense, 2021-03-15:
Okay, no problem.  I will keep this case open for a further time period.  Thanks for the update!

jimmek, 2021-05-19:
Hello Marty,

Quick question. When running Viewer with Stereo Module, RGB Camera, and Motion Module on, and then Record button used to start and end data collection. does Viewer write the IMU data (Linear Acc and Gyro data) to the bag file?

Thank you.

MartyG-RealSense, 2021-05-19:
Yes, the Viewer will record whatever streams are currently enabled to the bag when the Record option is activated.  

There is an example of a pair of such bags in a zip file called **Outdoor scene with D435i pre-production sample (Depth from Stereo with IMU)** on Intel's **Sample Data** page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

[image]

[image]

closed 2021-05-31 by MartyG-RealSense