realsenseai/librealsense #8481: Accessing the IMU data using the Python interface

bobdavies2000, 2021-03-02:
I have the latest firmware for the D455 and the latest version of librealsense installed. I am using the latest version of Windows 10. I am successfully running the example3 that is provided in the wrappers/TensorFlow directory. I tried to add capturing the IMU data by enabling the gyro and accel with the following statements:

config.enable_stream(rs.stream.gyro)
config.enable_stream(rs.stream.accel)

I wanted to read the gyro and accel data with the following:

gyroVal = frames.first_or_default(rs.stream.gyro, rs.format.xyz32f)
accelVal = frames.first_or_default(rs.stream.accel, rs.format.xyz32f)

However, these statements made the Python script fail. Remove them and everything is working fine. What should I have done? Is the IMU accessible in the Python interface?  Are there an examples of using the IMU interface from Python?  It is working fine in my Windows interface.

Bob Davies

MartyG-RealSense, 2021-03-03:
Hi @bobdavies2000   Yes, the IMU is accessible from Python.  A script in the link below for accessing and printing the gyro and accel values may be a useful reference for you.

https://github.com/IntelRealSense/librealsense/issues/3409#issuecomment-471673994

bobdavies2000, 2021-03-03:
Thanks for the link!  I will give that a try...

closed 2021-03-03 by bobdavies2000