realsenseai/librealsense #8482: libusb errors with L515

ThatGeoGuy, 2021-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515 | 
| Firmware Version                   | (01.04.01.00) | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-41-generic                                     | 
| Platform                                 | x86_64  |
| SDK Version                            | 2.42.0                        | 
| Language                            |  C                          | 
| Segment			|  other                       | 

### Issue Description

Occasionally I see some libusb errors followed by a SIGSEGV when trying to stream depth + IR from an L515.

Usually stuff like:

```
libusb: debug [libusb_get_config_descriptor] index 0
```
or
```
libusb: debug [parse_endpoint] skipping descriptor 30
```

I'm using the USB cable that came with the L515, and I've been plugging this directly into the back of my motherboard in a USB3.0 port. I had a suspicion that I've been passing something wrong into the C-API, but in such a case an exception _should_ be caught and an `*rs2_error` should be returned. The errors happen sporadically, and I can't seem to reliably reproduce them, but the same errors **do not** occur at all with my D435i.

ThatGeoGuy, 2021-03-03:
I opted to upgrade my firmware since it seemed out of date when running realsense-viewer.

Since then, I consistently am running into this error with the L515 as well:

```
xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
```

Any hints? Perhaps the librealsense2-dkim or udev rules shipped for Ubuntu 20.04 is incorrect somehow?

ThatGeoGuy, 2021-03-10:
So after a bit of trial-and-error, I realized that these errors were largely a result of trying to execute pipelines / contexts across multiple threads for the same device.

What guarantees can we assume about contexts / pipelines in a multi-threaded context?

RealSenseSupport, 2021-03-24:
Hi @ThatGeoGuy 

Are things running well for you now?  How about multi-threaded setup?

ThatGeoGuy, 2021-03-25:
I have mostly stuck to doing things in a single threaded manner, and they seem stable(ish?). What guarantees can we assume about contexts / pipelines in a multi-threaded environment?

RealSenseSupport, 2021-04-25:
Hi,

Not sure exactly what you mean by guarantees but there is some documentation on the subject here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md#high-level-pipeline-api

and here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md

that might be useful.

Thanks

closed 2021-07-02 by MartyG-RealSense