realsenseai/librealsense #8492: Python interface to IMU working?

bobdavies2000, 2021-03-03:
Using the latest version of librealsense and firmware on the latest Windows 10, the D455 or D435i can't seem to get the IMU data in Python - the images are returned fine.  The recommended method is the following Python script which was created about 2 years ago.  I have reset the camera and tried several times and it times out on the wait_for_frames().  All the Python examples distributed with the download for librealsense work fine but none of them use the IMU from Python.  Is there some trick to getting the following to work?  I am using the 64-bit Python 3.7 distribution.  Could it be the version of Python?

Bob Davies

import pyrealsense2 as rs
import numpy as np

def initialize_camera():
    # start the frames pipe
    p = rs.pipeline()
    conf = rs.config()
    conf.enable_stream(rs.stream.accel)
    conf.enable_stream(rs.stream.gyro)
    prof = p.start(conf)
    return p

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

p = initialize_camera()
try:
    while True:
        f = p.wait_for_frames()
        accel = accel_data(f[0].as_motion_frame().get_motion_data())
        gyro = gyro_data(f[1].as_motion_frame().get_motion_data())
        print("accelerometer: ", accel)
        print("gyro: ", gyro)

finally:
    p.stop()

MartyG-RealSense, 2021-03-04:
Hi @bobdavies2000   Is the above script able to function if you remove the line **return p** underneath your pipe start line?

bobdavies2000, 2021-03-04:
Marty - that return p is needed.  Instead I replaced the initialize_camera() call with the subroutine code so no return is needed.  The changed code fails in the wait_for_frames() as before.  There must be some larger environmental failing on my part - this code was working according to the post you provided.  I am running 64-bit Python 3.7.  Here is the minimalist version of the code.  If the indent fails to come through, everything after the while is indented once.  Note that the Python examples provided in the download are using something called "pipeline_wrapper".  Things changed since the original IMU example was written 2 years ago but none of the newer Python examples use the IMU.

Bob Davies 

import pyrealsense2 as rs
import numpy as np

p = rs.pipeline()
conf = rs.config()
conf.enable_stream(rs.stream.accel)
conf.enable_stream(rs.stream.gyro)
prof = p.start(conf)

while True:
    f = p.wait_for_frames()
    accel = np.asarray([accel.x, accel.y, accel.z])
    gyro = np.asarray([gyro.x, gyro.y, gyro.z])
    print("accelerometer: ", accel)
    print("gyro: ", gyro)

MartyG-RealSense, 2021-03-05:
Could you see if inserting a **try** instruction above your **while** instruction helps?

**try:
while True:**

bobdavies2000, 2021-03-05:
Marty - I added the try: and an except and the except clause gets the error "Frame didn't arrive within 5000" as the error message.  Does this test work on your system?  If you have Visual Studio installed with Python support and your system is working, it must be something I am doing wrong.  

Bob Davies

MartyG-RealSense, 2021-03-05:
I do not have the setup to run code tests.

There does not seem to be a problem with your own Python setup, given that you can run all the non-IMU examples without problems.

You could try the alternative Python IMU script in the link below:

https://github.com/IntelRealSense/librealsense/issues/3917#issuecomment-512336371

bobdavies2000, 2021-03-05:
Marty - thanks for your persistence in providing  links.  I believe that all of them work!  I tried the same script on another system where I had just installed Visual Studio (with Python 3.7) and the original example works fine.  So the problem is on my system and I will work on that.

closed 2021-03-05 by bobdavies2000

MartyG-RealSense, 2021-03-06:
You are very welcome.  Excellent to hear that you had success on another system @bobdavies2000 - thanks for the update!   :)