realsenseai/librealsense #8493: Reading temperature fails when in genlock mode

daziFR, 2021-03-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            | SDK 2.0 (build 2.41.0)                      | 
| Language                            |  C++                   | 
| Segment			|  others                           | 

### Issue Description
Trying to read temperature every few seconds in genlock mode (Inter Cam Sync Mode 4). From time to time, temperature read fails with hwmon_response::hwm_HWNotReady. 

Configuration:

- using hwm for temperature read 
`auto dbg = device.as<rs2::debug_protocol>();`
 `std::vector<uint8_t> cmd = {0x14, 0, 0xab, 0xcd, 0x2a, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};`
 `auto res = dbg.send_and_receive_raw_data(cmd);`

- auto exposure
- depth stream only

Issue mostly appears when there is a change in the scene brightness, so the auto exposure is busy calculating the right exposure.
I assume it is not recommended to use auto-exposure when in genlock mode (non-consistent frame rate).

So, I changed to manual exposure. And change exposure randomly before each temperature readout. Using this approach, issue appear more frequently.

Additionally,  when camera returns `hwmon_response::hwm_HWNotReady`, camera can be not responsive to hwm commands up to ~2.5s.

Is there some problem with resource sharing on the firmware side or is this expected behavior? When accessing projector temperature and changing exposure time at the same time (almost) introduces unknown behavior.

MartyG-RealSense, 2021-03-04:
Hi @daziFR   This problem is likely unrelated to genlock.  Issues with polling in C++ that did not occur in Python have been reported in the past.  A RealSense team member has said: "In C++ things tend to happen little faster (not to mention the extra layers of indirection).  It is likely that the first call made for the sensor temperature will arrive earlier than the sensor is ready to respond".

That information is sourced from a discussion in the link below:

https://github.com/IntelRealSense/librealsense/issues/6735

A RealSense user who was having problems with this issue solved it by reducing the frequency of temperature polling.

daziFR, 2021-03-04:
Thanks @MartyG-RealSense. Yes I read this topic #7389. But simply I cannot reproduce this in continuous mode. Maybe it is the same issue, but can be reproduced much easily in genlock mode.

Also, when using genlock in ROS Kinetic, this occurs occasionally, where the poll frequency is 10s, which is much slower than in my CPP example.

MartyG-RealSense, 2021-03-04:
Advice that I have from Intel is that genlock mode is not currently a mature feature, and so although you are free to use it and experiment with it, the original hardware sync system (mode 1 and 2) in the Multi-Camera Configurations white-paper document should be considered the mature 'validated' method of hardware syncing cameras.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

MartyG-RealSense, 2021-03-12:
Hi @daziFR   Bearing in mind the above statenent about genlock not being a mature system, do you require further assistance with this case please?   Thanks!

daziFR, 2021-03-12:
Hi @MartyG-RealSense,
There is no need for further assistance. Topic can be closed. I just wonder, are there some dates, timelines given for the new features (genlock, d455 synchronization etc) to be mature? Thanks.

MartyG-RealSense, 2021-03-12:
Thanks very much @daziFR for the update.  I do not have information about future plans for the mentioned features.

closed 2021-03-12 by MartyG-RealSense