realsenseai/librealsense #8496: my screen rotates 90 degrees automatically.

chrisxu1995, 2021-03-04:
Whenever I connect the USB of Intel realsense, my screen rotates 90 degrees automatically. Why is that? My system is Ubuntu20.04 , camera D435i.

MartyG-RealSense, 2021-03-04:
Hi @chrisxu1995  This may occur because your D435i's IMU component (which contains a gyro and accelerometer) is being detected by Ubuntu and activating Ubuntu's screen rotation feature.  You can lock screen rotation in Ubuntu, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/6941

chrisxu1995, 2021-03-04:
> Hi @chrisxu1995 This may occur because your D435i's IMU component (which contains a gyro and accelerometer) is being detected by Ubuntu and activating Ubuntu's screen rotation feature. You can lock screen rotation in Ubuntu, as described in the link below.
> 
> #6941

it works, thx

closed 2021-03-04 by chrisxu1995

jesusramondovale, 2024-05-29:
Hi! Same thing happens with my Ubuntu 22.04 LTS. 

I cannot turn off screen rotation because I need it rotated at 180º (inversed horizontal) 

Is there any way to disable this feature from Camera-Side? Thank you.