realsenseai/librealsense #8497: L515 firmware update fails with "ERROR: Recovery device did not connect in time!"

zhongfu, 2021-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 (pre-PRQ) | 
| Firmware Version                   | 01.03.04.201 | 
| Operating System & Version | Windows 10 20H2/Ubuntu 20.10 | 
| Kernel Version (Linux Only)    | 5.10.0-14-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.42.0 | 

### Issue Description
I've got a L515 here that refuses to enter recovery mode during firmware updates (through `rs-fw-update`, or RealSense Viewer 2.42.0 on Ubuntu 20.10 and versions 2.29.0 through 2.42.0 on Windows):

```
Started Firmware Update process
Backing-up camera flash memory
Backup completed and saved as '/home/.../.0003a84de99d.20210304_190338.bin'
Requesting to switch to recovery mode

ERROR: Recovery device did not connect in time!
```

(On Windows, the flash backup process proceeds much quicker than on Linux, but I'm not sure if that's relevant here since the firmware update fails anyway.)

At the same time, the following shows up repeatedly in `dmesg` -- rapidly (~19 lines/sec) while camera flash is being backed up, slowing down (~6 lines/sec) after requesting a switch into recovery mode, and stopping when the backup process fails:

```
[21089.868617] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[21089.920653] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[21089.972715] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

Running `rs-fw-update -l` after a failed firmware update returns the following:

```
$ rs-fw-update -l

connected devices:
RealSense error calling rs2_create_device(info_list:0x55ca07b8f910, index:0):
    set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 04/03 19:16:12,761 ERROR [139707912757376] (types.h:315) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
```

Depth/infrared/RGB image streams and motion data show up fine, but only before a firmware update is initiated. (A reconnect is required for normal (?) operation after a failed update attempt.)

Any ideas on what I can try next?

drewm1980, 2021-03-08:
This is impacting me as well. I'm stuck on L515 firmware 01.05.04.01. I've tried the firmware update both on Ubuntu 18.04 and recent Windows. Realsense-viewer 2.36.0 on Windows is able to flash the firmware, but doesn't do so to the latest version.

zhongfu, 2021-03-08:
> This is impacting me as well. Try Realsense-viewer 2.36.0 on Windows; that seems to have worked for me.

No dice either, unfortunately:

```
Started Firmware Update process
Backing-up camera flash memory
Backup completed and saved as 'C:\Users\...\AppData\Roaming\00000003a84de99d.20210308_204745.bin'
Requesting to switch to recovery mode
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor
Exception caught in FW Update process-flow: set_xu(ctrl=1) failed! Last Error: Bad file descriptor

ERROR: Recovery device did not connect in time!
```

RealSenseSupport, 2021-03-24:
Hi @zhongfu 

Are you able to show images from RealSense Viewer as to the camera info?

The initial information says that the L515 camera that you have is pre-PRQ with FW 01.03.04.201

zhongfu, 2021-03-25:
@RealSenseSupport here you go:

[image]

RealSenseSupport, 2021-03-29:
Thanks @zhongfu, we are looking into this.

closed 2021-03-29 by zhongfu