realsenseai/librealsense #8498: Realsense Error, frame didn't arrive within 15000 when rs-pcl-color.cpp is run multiple time

microbot7273, 2021-03-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.11.000 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-66 generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                           | 
| Language                            | c++                          | 
| Segment			|  Robot                        | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have been trying to run rs-pcl-color provided in the wrapper directory of the librealsense repository. After building the file, I tried running it and it worked perfectly. I was able to visualize. Once when the program is closed and again tried to rerun it, i get the following error saying  :

"""RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5582ab448d00):
    Frame didn't arrive within 15000
""""
This problem keeps persisting and the camera is responsive only on the first run after connecting it to  the PC. From the second run onwards this error persists. This error also persists when using realsense-viewer. Can you please provide me a solution ? Thank you.

MartyG-RealSense, 2021-03-04:
Hi @microbot7273   As a starting point in investigating your case, could you replicate the conditions that cause the RealSense Viewer to experience the **Frame didn't arrive within 15000** error.  Then go to the **More** option near the top of the Viewer's options side-panel and select the **Hardware Reset** option from its menu.

If this corrects the problrm then it would suggest that adding a **hardware_reset() instruction** to your script to reset the camera when the program runs may be worth trying.

microbot7273, 2021-03-04:
Hello , 

Yes hardware_reset() works for me. Although can you please clarify what the reasons might be behind this strange behaviour ? When i run the same code on ubuntu 16.04, this problem does not occur. 

Also right now, once I did a hardware reset in the viewer, it worked for me. Then I have turned both the stereo and RGB camera and then changed the resolution. When turned back on, this problem again occurs. This feels to me like the original problem where the camera pipeline only responds to the first connection and later on is irresponsive to the follow-up connections.

MartyG-RealSense, 2021-03-04:
Thanks very much for the information.  May I ask first that as well as enabling / disabling the Stereo Module and RGB, are you also activating the Motion Module in the Viewer at any time during the same session please?

microbot7273, 2021-03-04:
No , Just the rgb camera and stereo camera.

MartyG-RealSense, 2021-03-04:
Okay, thank you.  **Frame didn't arrive** errors have been known to also occur on Ubuntu 16.04, like the '5000' one in the link below.  Like in your case, a hardware reset corrected the problem.

https://github.com/IntelRealSense/librealsense/issues/1086

When the '15000' variety has occurred though, I tend to see it in cases involving multi-hour long runs of the camera rather than at pipeline start-up.

In the case in the link below where **Frame didn't arrive within 15000** was occurring from the start of streaming, updating the firmware of the camera corrected it.

https://github.com/IntelRealSense/librealsense/issues/4071

microbot7273, 2021-03-04:
okay Thank you. For now , I guess i will have to work with reset_hardware() method. Incase you find out the underlying cause or need more information, please let me know.

MartyG-RealSense, 2021-03-04:
I don't have any further suggestions to add at this time.  Good luck with your project!

MartyG-RealSense, 2021-03-12:
Hi @microbot7273   Do you require further assistance with this case, please?  Thanks!

microbot7273, 2021-03-12:
Hello @MartyG-RealSense , Thanks for asking. Though the above-mentioned problem is persistent, for the time being, I am following the routine you have suggested earlier. I believe, we can close this ticket for now. Thank you once again for the support.

closed 2021-03-12 by microbot7273

MartyG-RealSense, 2021-03-12:
You are very welcome @microbot7273 - thanks for the update!