realsenseai/librealsense #8500: D435 camera disconnected during update

echapp, 2021-03-04:
| | |
|---|---|
|**librealsense**|2.40.0 RELEASE|
|**OS**|Windows|
|**Camera Model**|D435|

Hello,

So, I messed up. Today, I launched the intel Realsense viewer, which was working fine before (a month or two ago), and it kept crashing. I figured that it was because it was only the 2.40 version, and not the 2.42 : they asked if I wanted to update, I said yes. An then, because it was still crashing, I disconnected the camera. But I didn't see that it was written not to disconnect the camera.
So, I know, I'm an idiot. Now, I've tried to solve the problem for the past hours, but it seems that I only made is worse. Here's what happened/what I did : 
- I closed everything out of panic (did I already said I was an idiot ?)
- After I disconnected, I tried to reconnect it. Windows 10 refused to recognize it, and now if telling me that it's an 'unknown USB" thing
- I tried to update the driver, downloading it from internet. Didn't work.
- I tried to update my intel librarie. As github is not really my cup of tea (uderstand : I don't understand anything about it, and would rather swim in an ice-cold sea to from France to England than having to download or update anything using it. I'm still working on that) I ended up re-installing the entire sdk, I think. The command I did was a "git clone". 

I'm ready to give you any informations you need, but I don't really know what you'll need. And yes, I'm on a USB3 port, that worked fine before. 

Again, I'm sorry for being like this. Can anyone help me making my camera work again like it used to do ? Also, if it involve downloading stuff from github, you may want to be very specific about the commands things I need to do. As I said, my github abilities is neer the jellyfish ones. 

I will thank the good souls that'll help me for a thousand year !

PS : I don't know how to put tags on issues I'm sorry

MartyG-RealSense, 2021-03-04:
Hi @echapp  It is a positive sign that your camera is able to be recognised.  

Have you tried updating the firmware in the Viewer by going to the **More** option near the top of the Viewer's side-panel and selecting the **Install Recommended Firmware** option, please?  If the update is able to be performed then the firmware should start updating automatically.

[image]

echapp, 2021-03-04:
Hello, thank you so much for the quick response.
I've tried that a few times, but get the same error everytime :

Started Firmware Update process
Recovery device connected, starting update

ERROR: UNKNOWN in rs2_update_firmware_cpp(device:0000026019638A50, fw_image:0000026020ADA060):
Device: 045323020527 failed to download firmware
Please verify that no other librealsense application is running

I'm now trying to update the firmware in an other way, as it seems that it's the problem.

Also, someone explained me how to update from github. It was not as hard as I tought !

MartyG-RealSense, 2021-03-04:
There was a past case of this error that apparently resulted from an **unsigned** firmware driver trying to be used with the camera.  If this is the cause of your problem, I am uncertain why the firmware file used by the **Recommended Firmware** option would be regarded as being unsigned.

Let's try a manual firmware installation.

1.  Please download the firmware file **Signed_Image_UVC_5_12_11_0.bin** from the firmware download page.

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

2.  Go to the **More** option of the Viewer again and this time select the menu option **Update Firmware**.  when the file selector box pops up, navigate to the location that you downloaded the firmware file to and select it.  The firmware update process should then commence.

echapp, 2021-03-04:
No, sorry there is always the same error ...

MartyG-RealSense, 2021-03-04:
If you left-clck on the 'Info' option at the top of the Viewer side-panel, which firmware version does it say is currently installed?

[image]

echapp, 2021-03-05:
[https://zupimages.net/up/21/09/ffcb.jpg](url)

Here is the only informations I have. It's not working anymore, I can't stream anything, and even with my code (which lauches streams) it's not working. 
Is there a way to reset the firmware or something like that ? It seems like I really messed it up, maybe corrupted it or something ...

MartyG-RealSense, 2021-03-05:
In most cases a camera that is in recovery mode is recoverable.

In case the failure to update is related to a corrupted calibration table inside the camera, let's try resetting its calibration to factory defaults in the RealSense Viewer.

Go to the **More** option and select the **Calibration Data** option from its menu.

[image]

On the pop-up window that appears, left-click the box next to **I know what I'm doing** to highlight it.

[image]

Highlighting this box causes an option called **Restore Factory** on the top of the pop-up window to become usable.  Left-click on it.  The window may briefly flash as it updates.

Then go back to the **More** menu and select **Install Recommendd Firmware** again to see if the firmware can now update.

echapp, 2021-03-05:
I can't access the whole menu from the viewer unfortunately (see picture)
https://zupimages.net/up/21/09/yk0i.jpg

I don't have much choise in the viewer, maybe there's an other tool I can use ?

MartyG-RealSense, 2021-03-05:
Aside from the Viewer, the other official method of updating firmware with the SDK is the tool **rs-fw-update**.  This can be run from the Windows command prompt interface (go to the text box at the bottom of the Windows screen, next to the corner button where you shut down, and type command into the box to find the Command Prompt).

[image]

The link below has instructions for using the recovery mode of rs-fwupdate.

https://github.com/IntelRealSense/librealsense/issues/7783#issuecomment-728079570

echapp, 2021-03-08:
It's working ! 
I had to do it twice, the first time it didn't work, but the second time, the recovery was done. Thank you very much for all the help you provided, and for being very clear in you explanations !

MartyG-RealSense, 2021-03-08:
You are very welcome @echapp - thanks very much for the updates!

closed 2021-03-14 by MartyG-RealSense