realsenseai/librealsense #8503: TypeError: argument 0 of Sensor.open() should be a/an StreamProfile

jameszw001, 2021-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  Linux version 5.4.0-65-generic (buildd@lgw01-amd64-048) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #73~18.04.1-Ubuntu SMP Tue Jan 19 09:02:24 UTC 2021 | 
| Platform                                 | PC|
| SDK Version                            |  { how to check this one?}                          | 
| Language                            |  {nodejs }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
I will add more above later.
I am trying the realsense_viewer example and I am getting:

`start command
{ tag: 'start',
  data: { sensor: 'RGB Camera', streams: [ [Object] ] } }
[ { stream: 'color',
    resolution: '1280*720',
    format: 'rgb8',
    fps: 30 } ]
open profiles:
[]
/home/a/Downloads/librealsense-master/wrappers/nodejs/index.js:3424
          throw new TypeError(wrongTypeErrMsgPrefix + expectedType.name);
          ^

TypeError: argument 0 of Sensor.open() should be a/an StreamProfile
    at checkArgumentType (/home/a/Downloads/librealsense-master/wrappers/nodejs/index.js:3424:17)
    at ColorSensor.open (/home/a/Downloads/librealsense-master/wrappers/nodejs/index.js:743:7)
    at Realsense._handleStart (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/index.js:310:14)
    at Realsense.processCommand (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/index.js:111:14)
    at WebSocket.cmdSocket.on (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/index.js:372:13)
    at WebSocket.emit (events.js:198:13)
    at Receiver.receiverOnMessage (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/node_modules/ws/lib/websocket.js:720:20)
    at Receiver.emit (events.js:198:13)
    at Receiver.dataMessage (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/node_modules/ws/lib/receiver.js:414:14)
    at Receiver.getData (/home/a/Downloads/librealsense-master/wrappers/nodejs/examples/realsense_viewer/node_modules/ws/lib/receiver.js:346:17)
`

MartyG-RealSense, 2021-03-05:
Hi @jameszw001   May I first ask a couple of questions please:

1.  Have you installed the RealSense SDK (librealsense) software on your Ubuntu-using computer?

https://github.com/IntelRealSense/librealsense/releases

2.  If you have installed librealsense for Ubuntu, what method did you use to install it?

3.  I see that you are using the **nodejs** language, and the error messages that you received make reference to js scripts.    So I wonder whether you have tried to run realsense-viewer as a nodejs program when that tool is actually written in **C++** language.

jameszw001, 2021-03-05:
1. Have you installed the RealSense SDK (librealsense) software on your Ubuntu-using computer?
I followed this to install the SDK: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Have you successfully built the nodejs wrapper?
```
cd wrappers/nodejs/examples
npm install
node nodejs-capture.js
```
I got 
[image]

If you have built it, have you tested it by running a simple example script?
Hope the above answered your questions.

What instructions did you use to launch this example, please?

`node index.js
`
also

`node -v
v10.24.0
`

MartyG-RealSense, 2021-03-06:
As the RealSense Viewer example is in a sub-folder of the Examples directory called **realsense-viewer**, had you navigated to that directory before launching the Viewer's index.cpp script?  The instructions would probably look like the ones below.

cd wrappers/nodejs/examples/realsense_viewer
npm install
node index.js

jameszw001, 2021-03-06:
yes that is what I did

MartyG-RealSense, 2021-03-06:
The process that you are using to launch **realsense_viewer** seems logically correct (though I am admittedly not a nodejs specialist).  So it may be worth going back to looking at the installation process for the nodejs wrapper.

The **npm install** method is considered to be the simplest installation method instead of building the wrapper from source, but it has had problems in the past for some RealSense users.  A user posted their own modified guide for using npm install though in October 2020 that they said was successful for them,

https://github.com/IntelRealSense/librealsense/issues/7439#issuecomment-708503008

MartyG-RealSense, 2021-03-12:
Hi @jameszw001  Do you require further assistance with this case, please?  Thanks!

jameszw001, 2021-03-12:
Thanks for the solution provided. I might try. I ended up using the python
wrapper, which is working fine now. Thanks. I guess you can close this
ticket for now.

MartyG-RealSense, 2021-03-18:
Excellent news that the Python wrapper provided a solution for you.  Thanks very much for your agreement to close the case, which I shall now do.  Feel free to re-open it at a future date if you need to.  Good luck!

closed 2021-03-18 by MartyG-RealSense