realsenseai/librealsense #8505: hr returned: HResult 0x80070490

dinasind, 2021-03-05:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435i|
|**Serial Number**|048122071781|
|**Firmware Version**|05.11.01.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|046523051503|
|**Firmware Update Id**|046523051503|

I bought a new RealSense depth camera D435i this week, and installed RealSense View tool this morning. But when I open the RealSense View tool(2.42.0) on Win10 with RealSense depth camera D435i connected, some errors log show as below:
```
17:26:50 [Info]  - librealsense version: 2.42.0
17:26:50 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Framebuffer size changed to 1366 x 705
17:26:50 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Window size changed to 1366 x 705
17:26:50 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Scale Factor is now 1
17:26:51 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 254)
17:26:51 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\ds5\ds5-device.cpp:548 - The IMU sensor is undefined for PID ad1 and imu_chip_id: �
17:26:51 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - ks_control->KsProperty( reinterpret_cast<PKSPROPERTY>(&node), sizeof(node), &description, sizeof(KSPROPERTY_DESCRIPTION), &bytes_received) returned: HResult 0x80070490: "找不到元素。"
17:26:52 [Info]  - Intel RealSense D400 was selected as a default device
17:26:53 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - hr returned: HResult 0x80070490: "找不到元素。"
```

how to fix the issue?

MartyG-RealSense, 2021-03-05:
Hi @dinasind   I note that the default firmware installed in your new D435i is **5.11.1.100** which dates back to February 2019.  SDK 2.42.0 recommends the recent firmware **5.12.11.0**.

If your D435i can be detected by the Viewer and appears in the options side-panel, could you try updating your camera's firmware please by left-clicking on the **More** option near the top of the side-panel and selecting **Install Recommended Firmware** from its drop-down menu?  This should then update the firmware automatically.

[image]

dinasind, 2021-03-08:
Hi @MartyG-RealSense,

Thanks for your reply.
I have updated the D435i firmware to Version 05.12.11.00

| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D430I|
|**Serial Number**|048122071781|
|**Firmware Version**|05.12.11.00|
|**Advanced Mode**|NO|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|046523051503|
|**Firmware Update Id**|046523051503|

but there is another error log:
_11:30:26 [Info] types.h:1103 - Right MIPI error_

All the logs show as below:
```
11:30:08 [Info]  - librealsense version: 2.42.0

11:30:08 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Framebuffer size changed to 1366 x 705
11:30:08 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Window size changed to 1366 x 705
11:30:08 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2588 - Scale Factor is now 1
11:30:09 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:381 - Found 1 RealSense devices (mask 254)
11:30:09 [Info]  - Intel RealSense D430I was selected as a default device
11:30:25 [Info]  - Starting streaming of Depth, Infrared...
11:30:25 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: Y8 Infrared, 
Resolved to: Y8 Infrared, 
11:30:25 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: Z16 Depth, 
Resolved to: Z16 Depth, 
11:30:26 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\types.h:1103 - Right MIPI error
11:30:26 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\notifications.cpp:514 - Right MIPI error
11:30:54 [Info]  - Starting streaming of Accel, Gyro...
11:30:54 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\ds5\ds5-motion.cpp:473 - IMU Calibration is not available, default intrinsic and extrinsic will be used.
11:30:54 [Error] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:179 - Undefined platform with IMU, use default intrinsic/extrinsic data, PID: 2891
11:30:54 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:194 - IMU extrinsic table not found; using CAD values
11:30:54 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:225 - Depth Module V2 IMU Accelno valid intrinsic available, use default values.
11:30:54 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:237 - Depth Module V2 IMU Gyrointrinsic not valid, use default values.
11:30:54 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: MOTION_XYZ32F Accel, 
Resolved to: MOTION_XYZ32F Accel, 
11:30:54 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:225 - Depth Module V2 IMU Accelno valid intrinsic available, use default values.
11:30:54 [Info] c:\jenkins_sys_rsbuild\workspace\lrs_windows_compile_pipeline\src\ds5\ds5-motion.h:237 - Depth Module V2 IMU Gyrointrinsic not valid, use default values.
11:30:54 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\sensor.cpp:1413 - Request: MOTION_XYZ32F Gyro, 
Resolved to: MOTION_XYZ32F Gyro, 
```
And there is no RGB menu on the RealSense View tool:
[image]

MartyG-RealSense, 2021-03-08:
The RealSense Viewer is identifying your D435i as a **D430i**.  The D430 model does not have an RGB sensor, which is why the RGB stream would be unavailable in the Viewer if it mis-detects that a D430 camera model is being used.

If the model is consistently being detected as D430i, could you try downgrading to the previous **5.12.10.0** firmware version please.  This can be done by following the steps below.

**STEP 1**
Download the firmware **.bin** file or 5.12.10.0 from the firmware releases page.

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

**STEP 2**
Go to the **More** menu in the Viewer side-panel and this time select the **Update Firmware** menu option.  A pop-up file selector window will appear.   Navigate to the location on your computer that you downloaded the .bin file to and select it.  The firmware update process should then begin.

If downgrading the firmware to 5.12.10.0 enables the camera to be successfully detected as D435i, you can then re-upgrade to firmware **5.12.11.0** if you wish.

dinasind, 2021-03-09:
Hi @MartyG-RealSense, 

I have downloaded  and updated the firmware V5.12.10.0 according to your steps. But unfortunately, the camera is still detected as D430i. Even I re-upgraded the camera to firmware 5.12.11.0,  it cannot fix the issue.

MartyG-RealSense, 2021-03-09:
Next, let's test whether resetting the camera to factory-new defaults makes a difference.

Please go back to the **More** menu and this time select the **Calibration Data** menu option.

[image]

When the Calibrati

dinasind, 2021-03-10:
I have done these operations as you commented, but that makes no difference to D430i status.

MartyG-RealSense, 2021-03-10:
Next, let's test whether the RGB sensor in your D435i is working, which may indicate whether or not the detection issue is in the RealSense SDK.

In the search box at the bottom of your Windows screen (next to the button in the corner where you shut down the computer), enter the word **camera**.  Select the app called **Camera**.  This is the Windows webcam app (not a RealSense camera app).

You may need to left-click on the **Change Camera** option after the program launches to make the D435i's RGB stream appear.

[image]

If you see the message below when switching the camera, ignore it and click **Close**.

[image]

dinasind, 2021-03-10:
It cannot start the D435i's RGB camera by the camera app on Win10.
[image]

how to fix it?

dinasind, 2021-03-10:
Hi @MartyG-RealSense,

I re-try to start the D435i's RGB camera again, there is D435i's RGB stream.
But the camera preview and picture don't seem normal.
[image]

MartyG-RealSense, 2021-03-10:
Could you right-click on the Depth driver please in the Device Manager and select **Uninstall Device** to remove it from the Device Manager.  Then unplug the camera from the USB port, wait a couple of seconds and plug it back in.  The camera should then reinstall the two drivers (RGB and Depth) automatically over an internet connection and they will then appear in the Device Manager.

dinasind, 2021-03-11:
Hi @MartyG-RealSense,

I have done these steps as you commeted, there still have no the RGB camera driver in the Device Manager.
And I will close this case as I have asked the vendor to exchange a new D435i for me.

Thanks again for all of your help!

closed 2021-03-11 by dinasind

MartyG-RealSense, 2021-03-11:
You are very welcome @dinasind - thanks very much for the update!