realsenseai/librealsense #8506: D415 no RGB data

ShettyHarapanahalli, 2021-03-05:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415|
|**Serial Number**|839112061189|
|**Firmware Version**|05.12.11.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|841113022746|
|**Firmware Update Id**|841113022746|

I am trying to view the RGB data with the python interface or with the realsense viewer I can not see the same in either.

MartyG-RealSense, 2021-03-05:
Hi @ShettyHarapanahalli   When viewing the RGB stream in the Viewer or in your own Python application, are you using the RGB stream only?  Or has the IMU stream also been enabled?

ShettyHarapanahalli, 2021-03-05:
I have enabled RGB only, I don't see the IMU as it's 415.

MartyG-RealSense, 2021-03-05:
Ah yes, sorry.  Do you receive any messages in the RealSense Viewer such as **No Frames Received** or is it just blank?

ShettyHarapanahalli, 2021-03-05:
Sharing the cmd line log - 
$ [4714:4718:0305/102049.977803:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/102051.615543:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/102519.871172:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/102519.911124:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/102949.881860:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/102951.687162:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/103419.982745:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/103420.187497:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/103849.813641:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100
[4714:4718:0305/103850.736231:ERROR:ssl_client_socket_impl.cc(962)] handshake failed; returned -1, SSL error code 1, net_error -100

suman@SumanDL:~/sandbox/teat/src$ realsense-viewer 
 05/03 10:40:18,693 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,693 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,693 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Backlight Compensation control for RGB Camera
 05/03 10:40:18,693 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,693 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,693 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Brightness control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Contrast control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Exposure control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Gamma control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Hue control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Saturation control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Sharpness control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add White Balance control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Enable Auto Exposure control for RGB Camera
 05/03 10:40:18,694 ERROR [139900858898176] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 10:40:18,694 ERROR [139901128079104] (sensor.cpp:523) acquire_power failed: failed to set power state
 05/03 10:40:18,694 WARNING [139901128079104] (sensor.cpp:1112) Failed to add Enable Auto White Balance control for RGB Camera
[4667:4694:0305/104102.003608:ERROR:connection_factory_impl.cc(428)] Failed to connect to MCS endpoint with error -106

MartyG-RealSense, 2021-03-05:
Is there any improvement if you go to the **More** option of the Viewer's side-panel and select **Hardware Reset** from its drop-down menu to reset the camera?

[image]

ShettyHarapanahalli, 2021-03-05:
the color image appears only when the depth is enabled.

closed 2021-03-05 by ShettyHarapanahalli

MartyG-RealSense, 2021-03-05:
Hi @ShettyHarapanahalli   I see that you closed this case - is the problem solved now, please?