realsenseai/librealsense #8507: Receiving two streams from one sensor

RisingMichael, 2021-03-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC |
| SDK Version                            |   2.  | 
| Language                            |  C++ | 
| Segment			|  Vision    | 

### Issue Description

I have a problem receiving the depth frame and IR frame from the depth sensor at the same time. I don't know if this problem has already been solved but I haven't found an answer to it yet.  
To enable the streams I used the algorithm provided by #6882.

    rs2::context ctx;
	rs2::syncer sync(2);
	rs2::device dev = ctx.query_devices()[0];
	rs2::sensor depth_sensor =   dev.query_sensors()[0];

	int depth_x = 848;
	int depth_y = 480;

	//setup grayscale stream as solved by https://github.com/IntelRealSense/librealsense/issues/6882
	std::unique_ptr<rs2::stream_profile> grayscale_stream = 
		enable_stream(depth_sensor.get_stream_profiles(),
			RS2_STREAM_INFRARED, depth_x, depth_y, RS2_FORMAT_Y8, 30);

	//setup depth stream  as solved by https://github.com/IntelRealSense/librealsense/issues/6882
	std::unique_ptr<rs2::stream_profile> depth_stream =
		enable_stream(depth_sensor.get_stream_profiles(),
			RS2_STREAM_DEPTH, depth_x, depth_y,	RS2_FORMAT_Z16, 30);
	depth_sensor.open( { *depth_stream, *grayscale_stream } );
	
	depth_sensor.start(sync);

	for (int i = 0; i < 10; i++)
	{
		rs2::frameset frames =  sync.wait_for_frames();

		rs2::depth_frame depth = frames.get_depth_frame();
		rs2::video_frame gray = frames.get_infrared_frame(1);
	}	

	depth_sensor.stop();
	depth_sensor.close();

The wait_for_frames() method provided by the syncer only returns a frameset holding either the depth frame OR the IR video-frame but not both at the same time. However I need both at the same time.
How can I achieve this?
Thank you.

MartyG-RealSense, 2021-03-06:
Hi @RisingMichael   Is the information in the link below helpful to you, please?

https://github.com/IntelRealSense/librealsense/issues/7010

RisingMichael, 2021-03-06:
> Hi @RisingMichael Is the information in the link below helpful to you, please?
> 
> #7010

Thank you for your reply. 
I think I found the problem: 

After testing it again and checking what frames are valid, I've found out that only the first couple (the exact number varies) of framesets don't include both frames.

	for (int i = 0; i < 10; i++)
	{
		rs2::frameset frames = sync.wait_for_frames();

		rs2::depth_frame depth = frames.get_depth_frame();
		rs2::video_frame gray = frames.get_infrared_frame();

		std::cout << "Frame number: " << i << '\n';
		std::cout << "Depth frame valid: " << (depth ? "true" : "false") << '\n';
		std::cout << "Gray frame valid: " << (gray ? "true" : "false") << '\n';

		std::cout << '\n';
	}	

This had the following output: 

          Frame number: 0
          Depth frame valid: true
          Gray frame valid: false
          
          Frame number: 1
          Depth frame valid: false
          Gray frame valid: true
          
          Frame number: 2
          Depth frame valid: false
          Gray frame valid: true
          
          Frame number: 3
          Depth frame valid: true
          Gray frame valid: false
          
          Frame number: 4
          Depth frame valid: true
          Gray frame valid: true
          
          Frame number: 5
          Depth frame valid: true
          Gray frame valid: true
          
          Frame number: 6
          Depth frame valid: true
          Gray frame valid: true
          
          Frame number: 7
          Depth frame valid: true
          Gray frame valid: true
          
          Frame number: 8
          Depth frame valid: true
          Gray frame valid: true
          
          Frame number: 9
          Depth frame valid: true
          Gray frame valid: true

closed 2021-03-06 by RisingMichael

MartyG-RealSense, 2021-03-06:
Thanks very much for the update @RisingMichael 

It is possible to skip the first several frames after the pipeline starts using a **for (int i)** instruction.  It looks as though you are already using that method with **for (int i = 0; i < 10; i++)**

Thanks again!