realsenseai/librealsense #8508: Wrong distances when using align to color

AJMena, 2021-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10 }  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.41.0.2657 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
We did a program to measure some distance like the measure example but in 2D. The user sees the color image and can click to measure. We used the align to color so the image has larger resolution and the holes are not displayed. We recorded the depth at 848x480 and the color at 1280x720. Our problem is that the results are incorrect, the distances are lower than the real values. We tested changing the align to depth, in that case the results are correct.

So, is this behavier expected? Or maybe is this a different resolution problem?
Is there anything we can do to use align to color and measure correctly?

MartyG-RealSense, 2021-03-05:
Hi @AJMena   The SDK's **align processing block** makes automatic adjustments for factors such as different stream resolutions.   So it is not something to be overly concerned about in my opinion. 

Using color to depth is not a wrong approach.  As you noted, the **rs-measure** example uses it.  My understanding though is that one of the reasons for it being based on color-to-depth was simply to provide a code example in the SDK of color-to-depth alignment, rather than because it was a better method of measurement.  A RealSense user who created a Python conversion of rs-measure demonstrated this by using depth-to-color in their version instead.

Depth to color alignment also has the advantage of the color being able to be subtracted and re-added in real-time as the draw distance of the depth data is changed, as demonstrated by the **rs-align-advanced** example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced

I am not personally aware of a difference in accuracy when using color-to-depth instead of depth-to-color.  I would unreservedly recommending using depth-to-color though, as there is a huge amount of evidence spanning years that it works, whereas the number of examples of color to depth that can be referred to are far smaller.

MartyG-RealSense, 2021-03-12:
Hi @AJMena  Do you require further assistance with this case, please?  Thanks!

AJMena, 2021-03-17:
No, thank you.
We finally opted for using align to depth for both frames. For better outcome we applied the filters to the depth frame before the alignment. Among the filters we used there was the hole filling filter. With that the resulting picture does not show holes, except in the borders. That way the result is accurate and good looking.
I am sorry for the late response.

closed 2021-03-17 by AJMena

MartyG-RealSense, 2021-03-17:
No problem at all @AJMena - it is great to hear that you achieved a solution that satisifes your needs.  Thanks for the update and the sharing of details of your method!

Applying post-processing **before** depth-to-color alignment is the recommended course of action, as it helps to reduce **aliasing** (jagged lines).