realsenseai/librealsense #8510: Issue with Patching Kernel Modules

yihang314, 2021-03-05:
When I tried to run `./scripts/patch-realsense-ubuntu-lts.sh` , I always get the following error `/home/edu/5.4.0-66-generic-videobuf2-common.ko': No such file or directory`

I tried both Ubuntu 20.04 LTS and 18.04 LTS, kernel versions are 5.4.

I searched around and I don't see anyone else has the same problem, any advice?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515                                                        | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)  | 
| Operating System & Version |   Ubuntu 18&20                                      | 
| Kernel Version (Linux Only)    |  5.4                                                         | 
| Platform                                 | PC: VirtualBox&  VMware Workstation  |
| SDK Version                            | 2.0                                                         | 
| Language                               |  C++                                                      | 
| Segment			         |  Robot                                                   |

EVMakers, 2021-03-06:
Same here. build on 18.0.5 LTS. 5.4.0-66-generic-videobuf2-common.ko': No such file or directory error when running ./scripts/patch-realsense-ubuntu-lts.sh

RealSenseSupport, 2021-03-09:
There is discussion at the following thread:  https://github.com/IntelRealSense/librealsense/issues/8513

seyoung-solteq, 2021-03-22:
Same problem. https://github.com/IntelRealSense/librealsense/issues/8125#issuecomment-802841334

RealSenseSupport, 2021-03-23:
Hi @yihang314 

Are you able to get past patching kernel modules?

closed 2021-06-01 by RealSenseSupport