realsenseai/librealsense #8514: The depth start of get_distance() function ?

gywhitel, 2021-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 

### Issue Description
When I use `get_distance()` to get the distance from camera to the surface in its front. There is always about 4mm error if I set the depth start at the imager, so I guess the function return the distance to front cover glass.
Is that so ? The depth starts from front cover glass ?

MartyG-RealSense, 2021-03-06:
Hi @gywhitel   Yes, on the cased versions of RealSense 400 Series cameras, depth is measured from the front glass.  On the caeless Depth Kit versions of RealSense cameras, depth is measured from the lens or from the back of the module.

You can get the **ground truth** of measurements by adding a minus value to them.  On the D435 and D435i models, that is **-4.2** mm

[image]

gywhitel, 2021-03-06:
Thanks ! Problem solved.

closed 2021-03-06 by gywhitel

MartyG-RealSense, 2021-03-06:
Thanks very much @gywhitel - I'm pleased I could help.  I have updated the table graphic to show the value for the D455 model too.