realsenseai/librealsense #8515: RGB Camera resolution on Realsense viewer is different between Windows and Linux

takmin, 2021-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 |
| Firmware Version                   | 01.05.04.01 | 
| Operating System & Version |   Win 10 / Ubuntu 18   | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0-91)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0                      | 
| Language                            |  {C }                          | 
| Segment			|  others }                          | 

### Issue Description
On realsense viewer, I can NOT choose 640 x 480 resolution of RGB camera on Windows 10.
Otherwise, I can choose it on Ubuntu.
The version of SDK and viewers are same.

How can I choose the resolution on Windows?

[image]

takmin, 2021-03-06:
On ubuntu, I can NOT choose RGB camera resolution "640 x 480" after I reconnected USB cable.
It is very strange behavior,  but problem is solved.

RealSenseSupport, 2021-03-09:
Supported RGB resolutions for L515 are as follows:  960x540, 1280x720, 1920x1080

Glad to see things are resolved.

closed 2021-03-09 by RealSenseSupport