realsenseai/librealsense #8516: Recording doesn't work properly

MatsalVistry, 2021-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435I } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.31.0) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | Jetson TX2 |
| SDK Version                            |  { 2.31.0 }                          | 
| Language                            |  {Python/OpenCV }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When attempting to record a video, the first few frames seem to be good and then it keeps sending the same frame.

The script im using to record:

```
while (time.time()-t0) <timeRecord:
    frames = pipeline.wait_for_frames()     
    color_frame = frames.get_color_frame()  

    if not color_frame:
        print("INVALID FRAME, SKIPPING")
        continue

    color_image = np.asanyarray(color_frame.get_data())
    framesList.append(color_image)
```

Any ideas why this is happening?

MartyG-RealSense, 2021-03-06:
The behavior of the script and the use of **append** reminds me of an old Python case where the script could not retrieve more than 15 frames because the frames were being saved into a **collection**.

https://github.com/IntelRealSense/librealsense/issues/946

The script in the link below may provide useful guidance about an alternative approach of recording to a numpy array.

https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267

MatsalVistry, 2021-03-06:
I came across that same case and found it similar to mine, thanks for the alternative approach!

closed 2021-03-06 by MatsalVistry

MartyG-RealSense, 2021-03-06:
Great to hear that the alternative worked for you @MatsalVistry  - thanks for the update!