realsenseai/librealsense #8520: error: cannot run gitproxy when running ./scripts/patch-realsense-ubuntu-lts.sh

xsqiao, 2021-03-08:
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-xenial-hwe folder

cloning to './ubuntu-xenial-hwe'...
error: cannot run gitproxy: no such file or directory
fatal: cannot start proxy gitproxy

MartyG-RealSense, 2021-03-08:
Hi @xsqiao   There is almost no information available about using librealsense with gitproxy.  I did find a single case though about adding the gitproxy protocol when trying to clone to ubuntu-xenial that may be helpful.  Please scroll down to the section titled **Problem 1 network proxy problem**.  

https://www.daimajiaoliu.com/daima/47617b2a8900410

xsqiao, 2021-03-09:
Thank you very much for your reply!

closed 2021-03-09 by xsqiao