realsenseai/librealsense #8521: Global time domain for RS2_FRAME_METADATA_SENSOR_TIMESTAMP

MBSadeghzadeh, 2021-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | () | 
| Operating System & Version |   {Win (10) |                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.38.1                          | 
| Language                            |  {C++}}                          | 

Realsense provides different kinds of timestamps. The most important one to me is `RS2_FRAME_METADATA_SENSOR_TIMESTAMP ` which is the timestamp of the middle of sensor's exposure calculated by device. I have enabled `Frame Metadata` so I am able to read all kinds of timestamps that each frame includes. 

As it has been written [here](https://intelrealsense.github.io/librealsense/doxygen/rs__frame_8h.html#a55750afe3461ea7748fbb2ef6fb19e8a), I wonder if `rs2_timestamp_domain` only specifies the clock in relation to which   `RS2_FRAME_METADATA_FRAME_TIMESTAMP ` was measured  or it also specifies the clock domain of ` RS2_FRAME_METADATA_SENSOR_TIMESTAMP` ?

rafaelspring, 2021-03-08:
Not the exact same question but lots of answers in this thread. Does this help?
#6433

MartyG-RealSense, 2021-03-08:
Thanks very much @rafaelspring for your assistance to @MBSadeghzadeh   :)

rafaelspring, 2021-03-08:
@MartyG-RealSense  Always happy to help :)

MBSadeghzadeh, 2021-03-08:
I hoped to understand this part of document. It was not pointed out in the other issue too.
There may be an ambiguity in the document as it says _**rs2_timestamp_domain** specifies the clock in relation to which frame timestamp was measured._ **but what is the frame timestamp?** Does it mean `RS2_FRAME_METADATA_FRAME_TIMESTAMP` or  it means other time stamps (`RS2_FRAME_METADATA_SENSOR_TIMESTAMP`) as well?

MartyG-RealSense, 2021-03-08:
@MBSadeghzadeh   Advanced mechanics of metadata is not one of my specialist knowledge areas, so I would refer you to an excellent source of information about timestamps in the link below.

https://github.com/IntelRealSense/librealsense/issues/2188

In this discussion, **frame_timestamp** is described as generated in firmware and designating the beginning of UVC frame transmission (the first USB chunk sent towards host).

MBSadeghzadeh, 2021-03-09:
You were right @MartyG-RealSense. I have done some tests and saw the same results. Global Time only applies to `framset.get_timestamp()`. I would prefer it applies to `Frame Metadata` though.

MartyG-RealSense, 2021-03-09:
Do you need advice about anything else please?  Thanks!

MBSadeghzadeh, 2021-03-09:
If you don't mind, I want to suggest that you consider the global time domain for `Frame Metadata` timestamps such as `RS2_FRAME_METADATA_SENSOR_TIMESTAMP` please. 
I also want to know more about global time. How is it calculated? what kind of protocol is used? how much is it precise and trustful?

MBSadeghzadeh, 2021-03-09:
I mean we can only see the frame timestamp in global time domain and other timestamps are in hardware clock domain. Is there any possibility to convert them to global time domain?

MartyG-RealSense, 2021-03-09:
Thanks very much @MBSadeghzadeh for the clarification.

@ev-mp  Would it be possible to provide advice about this question please:

**We can only see the frame timestamp in global time domain and other timestamps are in hardware clock domain.  Is there any possibility to convert them to global time domain?**

MBSadeghzadeh, 2021-03-13:
I have two D435i, if I made use of Hardware Synchronization and Global Time, would I have a zero time difference between two cameras' frame_timestamps ?

MartyG-RealSense, 2021-03-13:
@MBSadeghzadeh  If hardware sync is enabled then timestamps will drift apart over time.  If hardware sync is not enabled then the timestamps appear to be perfectly aligned.  The section of Intel's multiple camera white paper linked to below describes this, in the paragraph that begins with the bolded text **Now to the somewhat counter intuitive aspect of time stamps**.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-3-multi-camera-programming

MBSadeghzadeh, 2021-03-13:
How can I check timestamps drift? do I have to subtract consecutive timestamps of a single camera or subtract the timestamps of both cameras?

MartyG-RealSense, 2021-03-13:
You can observe them manually (for example, taking screenshots of both cameras' timestamps and comparing the images).

A hardware sync discussion in the link below includes a C++ script where drift is checked with code.

https://github.com/IntelRealSense/librealsense/issues/2637

[image]

Another way of validating sync that is suggested by the hardware sync white-paper is the use of high-speed LED panels.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-g-hw-sync-validation

https://support.intelrealsense.com/hc/en-us/community/posts/360049401673/comments/360012819533

MBSadeghzadeh, 2021-03-13:
Thanks @MartyG-RealSense. Reaching a zero time difference between two cameras is also possible but only with a bit coding.

closed 2021-03-13 by MBSadeghzadeh